OpenC3/cosmos #560: Script runner hash table limitations

MTI-twalker, 2023-03-26:
**Describe the bug**
A hash table subkey with multiple 2+ subkeys appears to cause some wierd error. 

**To Reproduce**
Run this script in scriptrunner
```
puts limits = {
  'val' => {
    'val Voltage' => {
      'tlm_name'        => 'target packet item', 
      'metric'          => 'V',
      'requirement_id'  => nil,
      'off_limits'      => {
          'yellow_lower'  => nil, 
          'red_lower'     => nil
      },
      'on_limits'       => {
          'red_lower'     => 1, 
          'yellow_lower'  => 1, 
          'yellow_upper'  => 1, 
          'red_upper'     => 1
      }
    }
  }
}
```

**Expected behavior**
I believe this is a valid hash table. 

**Screenshots**
Here is a capture of the log messages:
```
2023/03/26 05:05:46.533 (SCRIPTRUNNER): Exception in Control Statement - Script stopped: 2023_03_26_01_04_52_105_temp.rb
      ^
      }
__TEMP__/2023_03_26_01_04_52_105_temp.rb:16: syntax error, unexpected '}', expecting end-of-input
          ^~~~~~~~~~~~~~~~~~~~
          'red_upper'     => 1 end; RunningScript.instance....
__TEMP__/2023_03_26_01_04_52_105_temp.rb:15: void value expression
                             ^
          'yellow_upper'  => 1, 
__TEMP__/2023_03_26_01_04_52_105_temp.rb:14: syntax error, unexpected integer literal, expecting local variable or method
                    ^
          'red_lower'     => 1, 
__TEMP__/2023_03_26_01_04_52_105_temp.rb:12: syntax error, unexpected terminator, expecting literal content or tSTRING_DBEG or tSTRING_DVAR or tLABEL_END
...                              ^
...; __return_val = begin;       },
__TEMP__/2023_03_26_01_04_52_105_temp.rb:10: syntax error, unexpected '}', expecting `end'
^~~~~~~~~~~~
__return_val = nil; begin; RunningScript....
2023/03/26 05:05:46.531 (running_script.rb:1183): SyntaxError : __TEMP__/2023_03_26_01_04_52_105_temp.rb:10: syntax error, unexpected local variable or method, expecting '}'
2023/03/26 05:05:46.479 (SCRIPTRUNNER): Starting script: 2023_03_26_01_04_52_105_temp.rb
2023/03/26 05:05:46.380 (SCRIPTRUNNER): Script config/DEFAULT/targets/__TEMP__/2023_03_26_01_04_52_105_temp.rb spawned in 1.752890942 seconds <ruby 3.1.3>
```

**Environment (please complete the following information):**

- OS: windows 10
- OpenC3 COSMOS Version 5.5.2
- Browser
  -- [] Internet Explorer (IE)
  -- [x] Google Chrome
  -- [] Chromium based browsers
  -- [] Mozilla Firefox
  -- [] Safari

MTI-twalker, 2023-03-26:
Follow up, was having issues with docker yesterday. Was able to pull up a cli irb instance today and there were no issues with this hash table. It seems as though it is a script runner specific issue.

ryanmelt, 2023-03-27:
Sometimes the code instrumenter gets confused when you break up arrays, or hashes like this.

If you do Script -> Instrumented Script, you can see where it thinks it can add in some instrumentation in the middle, but it breaks the Hash.

In this case it works if you add a comma at the end of line 9, or move the closing bracket on line 10 up to line 9.

I'll leave this as a bug and see if we can make the instrumenter smarter.

MTI-twalker, 2023-03-27:
I am definitely ok with moving the closing bracket up. I didnt see any other way to format it to fix it. I did notice if I removed one of the subkeys it would work sometimes. I am not totally sure I know what instrumented script is either so i can look into that a bit more.

PR #568 (2023-04-03): Improve instrumentation
closes #560
closed 2023-04-03 by ryanmelt