OpenC3/cosmos #564: Is there a way for a protocol to derive if it is READ, WRITE, or both?

GTI-Dakota, 2023-03-28:
Hello,

after reviewing [this protocol file](https://github.com/OpenC3/cosmos/blob/main/openc3/lib/openc3/interfaces/protocols/protocol.rb), I don't see how a protocol can understand if it has read_allowed and write_allowed, like an interface does. Is there a sneaky clever way to do this where the protocol can derive this, independent of using a get_interface call and parsing that? For now, I have read/write args in my constructor to do this, but I don't feel its as clean and it adds 2 extra args
[image]

Thanks,

Dakota

ryanmelt, 2023-03-28:
Protocols have access to their associated interface.
```
@interface.read_allowed?
@interface.write_allowed?
```

GTI-Dakota, 2023-03-28:
Oh, good to know. Thanks!

GTI-Dakota, 2023-03-28:
This didn't work in the initialize() protocol function, unfortunately. 

the code:
[image]

The error when loading the gem
> _protocol.rb:53:in `initialize': undefined method `write_allowed?' for nil:NilClass (NoMethodError)

GTI-Dakota, 2023-03-28:
It works in the connect_reset, so I have reason to believe we are able to reference @interface if the protocol has been intialized.

ryanmelt, 2023-03-28:
Not going to work in initialize.  Should work everywhere else.

closed 2023-03-28 by GTI-Dakota