OpenC3/cosmos #566: ROUTE_PREFIX Question

ajavorskidev, 2023-03-28:
Hello, 
I'm currently attempting to utilize the new feature of dynamic routes through Traefik, and figured out that I could use `ROUTE_PREFIX` to expose my microservice API. Unfortunately other than setting the `ROUTE_PREFIX` in the plugin.txt file, I'm not too sure what else I am supposed to do to successfully be able to use this feature. I've read through the documentation on the main page, but it just mentions to set a unique prefix. Any guidance or help would be much appreciated!

ryanmelt, 2023-03-29:
ROUTE_PREFIX adds path to our traefik load balancer so you can access microservices external to COSMOS.  So if i added a route prefix of /myapi to a custom microservice, it would make that http endpoint available externally at http://localhost:2900/myapi

Note: It does also require specifying a PORT to work.

ajavorskidev, 2023-03-29:
Ah alright, so basically I could for instance access my routes in a microservice using ie. localhost:2900/myapi/route right?

jmthomas, 2023-04-12:
Assuming your ROUTE_PREFIX was `myapi/route`. Here's the code where we apply the prefix: https://github.com/OpenC3/cosmos/blob/main/openc3-cosmos-cmd-tlm-api/app/controllers/microservices_controller.rb#L40

closed 2023-04-12 by jmthomas