OpenC3/cosmos #567: Justify LABELVALUE widget

tsjordan-eng, 2023-03-29:
**Describe the solution you'd like**
Cosmos 4 LABELVALUEs justified the label to the left and the value to the right giving a clean, aligned look to all the fields. Now They're justified left so the alignment is different depending on the label length.

Can we go back to the right/left justification?

tsjordan-eng, 2023-05-03:
I'm not sure this is working in 5.6.1:
This is what I get:

[image]

But I expected more alignment (mocked up in MS Paint):
[image]

Note that the text boxes are aligned right while the label is left. Also the buttons at the bottom are spread out and fille the space (I drew them off-center on accident, but you get the idea).

tsjordan-eng, 2023-05-03:
oh... nevermind!
Is there a container tag I can pull that has those changes, or do I have to build the containers myself on the latest commit?

jmthomas, 2023-05-03:
You'll have to git clone the main repo and `./openc3.sh start` to rebuild and start everything if you want to test it

PR #635 (2023-04-24): Justify telemetry viewer widgets
closes #567