OpenC3/cosmos #580: Websocket interface

bradenvirdellgenevatech, 2023-04-03:
Hello,

Was wondering if the TCP interface has the capability to be used as a websocket instead of just a TCP connection?  If there is one could you point me to it and how to access it. From the documentation I have read I do not currently see one. The closest thing I have seen is a websocket stream but I do not know exactly how to access it or if I should.

Thanks,
Braden Virdell

ryanmelt, 2023-04-03:
I built the Websocket stream to support our websocket apis, and it could be pretty easily turned into an interface as well.  Haven't done that yet though.  I'll plan on working on this sometime after I implement the HttpInterface.  Hopefully by end of month.

bradenvirdellgenevatech, 2023-04-03:
Sounds good

ryanmelt, 2023-04-03:
Reopening to capture the future websocket interface.

bradenvirdellgenevatech, 2023-12-29:
Here is a simple websocket client,
It can connect to 1 ws or wss at a time and talk back and forth relatively well.  I needed to use the faye gem mainly due to it being the easiest websocket service I found with ruby. I have been using it for a few weeks now and haven't seen any major issues beyond the connection dying if it and the client haven't talked in a while. You can use this as a base if you wish, I'm sure there is a lot of improvement to be had.

`require 'json'
require 'faye/websocket'
require 'openc3/api/api'
require 'openc3/interfaces/interface'

DISCONNECT = "disconnect-service"

module OpenC3
    class WebsocketClientInterface < Interface
        def initialize(url)
            super()
            @url = url
            @client = nil
            @read_queue = Queue.new
            @recieve_thread = nil
            @connected = false
        end
        
        def connect
            @connected = false
            @recieve_thread = Thread.new { @client = self.eventmachine_websocket}
            @connected = true
            @read_queue.clear()
            @connected
            super()
        end

        def connected?
            @connected
        end

        def disconnect
            @read_queue << DISCONNECT
            @client.close
            @recieve_thread.kill.join
            @client = nil
            @connected = false
            @connected
            super()
        end

        def read_interface
            begin
                data = @read_queue.pop
                if data != DISCONNECT
                    if data.nil? or (data.length <= 0)
                        return :STOP
                    else
                        read_interface_base(data)
                        return data
                    end
                end
            rescue Exception => exception
                Logger.error("#{exception.message}: #{exception.backtrace}")
            end
        end

        def write_interface(data)
            begin
                write_interface_base(data)
                @client.send(data)
                data
            rescue
                Logger.error("in write interface rescue")
            end
        end

        def eventmachine_websocket()
            EM.run do
                @client = Faye::WebSocket::Client.new(@url)
              
                @client.on :open do |event|
                    Logger.info("Client been opened to #{@url}")
                end
              
                @client.on :close do |event|
                    Logger.info("#{[:close, event.code, event.reason]}")
                end
              
                @client.on :message do |event|
                  @read_queue << event.data
                end
            end
        end
    end
end`