OpenC3/cosmos #582: Protocol Reduce Packets

tsjordan-eng, 2023-04-04:
My packets have a grouping feature where a single packet can be split across multiple packets. The information to put it back together is in the header of each packet. I was wondering how we can do this in the protocol stage?

I found that some of the protocal files have a "reduce*" function, but it's not documented and I don't know what calls it. Am I on the right track? 
https://github.com/OpenC3/cosmos/blob/main/openc3/lib/openc3/interfaces/protocols/terminated_protocol.rb#L71
https://github.com/OpenC3/cosmos/blob/main/openc3/lib/openc3/interfaces/protocols/preidentified_protocol.rb#L132 

The documentation mentions caching data/packets and I see that some protocols have member variables like `@data`, but I don't see where they're assigned. I'm thinking that I need to write a custom protocol that caches the each packet in a group into a member variable, returning `:STOP` until the last packet of the group where I return the entire packet. 

Am I on the right track?

ryanmelt, 2023-04-04:
The documentation mentions caching data/packets and I see that some
protocols have member variables like @data, but I don't see where they're
assigned. I'm thinking that I need to write a custom protocol that caches
the each packet in a group into a member variable, returning :STOP until
the last packet of the group where I return the entire packet.

Am I on the right track?

Yes exactly.  You'll need to cache the data and return :STOP until you have
everything you need.

tsjordan-eng, 2023-04-05:
Okay. And the `data` is fresh data every time, right? Not the previous data plus new data?

I also don't understand[ this statement](https://openc3.com/docs/v5/protocols#custom-protocols) that "[an empty string] is a signal to Protocols that have cached up more than one packet worth of data to output those cached packets before any new data is read from the Interface." How does the interface know to send an empty string? What if this interrupts my "group" of packets? Do I just through it out and start over?

tsjordan-eng, 2023-04-06:
I think I found the answer to my question. If you are splitting packets instead of combining/reducing them, the empty string that the interface passes in allows your protocol to dispense each split packet one at a time without piling up the input packets. 

Maybe I'll add some verbiage to the documentation to explain splitting packets.

Also, `data` seems to be fresh on each call of `read_packet`.