OpenC3/cosmos #583: Monospace font where appropriate

tsjordan-eng, 2023-04-05:
The GUI might look better and be more functional with monospace font in select places

* Plugin.txt editor (I didn't realize it was editable until recently. Maybe use the script-runner editor here?)
* Plugin output
* Command sender History at the bottom
* Log messages, especially stderr and stdout where lines need to match up.
* generally anywhere there is code or config or stdout

@jmthomas, we spoke about this

ryanmelt, 2023-04-05:
@jmthomas Already did this in several places.  See https://github.com/OpenC3/cosmos/pull/574

jmthomas, 2023-04-06:
See #569 where we'll implement plugin.txt logic to force a selection as well as add the ACE editor.

However, there are a number of other places where I agree we should use monospace fonts. All the Admin tabs where we view configuration as well as the Redis tab.

PR #608 (2023-04-15): Ace edit json dialog and redis
closes #583
closed 2023-04-15 by jmthomas