OpenC3/cosmos #585: Is there a way to reference/get the connected clients list for an interface?

GTI-Dakota, 2023-04-05:
Hello,

When a client connects, I would like to do something specifically for that client from a TCP Server Interface (Or maybe from a protocol)
If possible, I would like to know how to derive the connection/file handle, or the client info.

My thoughts were doing the following:
 - Do something on client connect
 - Do something to/with a client connection when a protocol receives a packet/command (I.E. client says "echo hi" to just me)
 
 Is there an override you guys can expose in the future if this is currently not possible?
 
 Thanks,

Dakota

ryanmelt, 2023-04-05:
You'll need to do this with a custom interface (or protocol).   The tcpip_server_interface already has hooks for callbacks on connects.  See:

```
    # Callback method to call when a new client connects to the write port.
    # This method will be called with the Interface as the only argument.
    attr_accessor :write_connection_callback
    # Callback method to call when a new client connects to the read port.
    # This method will be called with the Interface as the only argument.
    attr_accessor :read_connection_callback
```

GTI-Dakota, 2023-04-05:
Brilliant. I will start with that. Thanks!

closed 2023-04-14 by jmthomas