OpenC3/cosmos #593: What's the correct calculation for the buffer depth?

GTI-Dakota, 2023-04-07:
Hello,

In order to avoid seeing the `Log writer out of order time detected (increase buffer depth?)` message in the logger, I am attempting to update my buffer depth in all of my gems to ensure that they do not spawn this error anymore. It's not listed in the documentation on how to calculate the buffer depth based on number of packets, would it be num_tlm_packets * 2?

Here is my plugin's target configuration. We retain logs for 1 week, rolling

> VARIABLE log_retain_time 604800
>
> TARGET MY_TARGET <%= my_target_name %>
>     CMD_DECOM_LOG_RETAIN_TIME <%= log_retain_time %>
>     TLM_LOG_RETAIN_TIME <%= log_retain_time %>
>     TLM_DECOM_LOG_RETAIN_TIME <%= log_retain_time %>
>     REDUCED_MINUTE_LOG_RETAIN_TIME <%= log_retain_time %>
>     REDUCED_DAY_LOG_RETAIN_TIME <%= log_retain_time %>
>     LOG_RETAIN_TIME <%= log_retain_time %>
>     REDUCED_LOG_RETAIN_TIME <%= log_retain_time %>

(Spacing is not playing nice with the quote)

Thanks,

Dakota

jmthomas, 2023-04-12:
The buffer depth value is difficult because it's all about the volume and variability of the data. We set the default for commands to 5 and telemetry to 60. This means the logger will buffer up this number of packets before decommutating them or writing them to the log file. Since each interface is independent now, packets can get out of order because they're time tagged at the interface and then put on a topic which is read by the decommutator and logger processes.

I'm not sure the best general rule to apply. I think the bare minimum is to take the total number of interfaces and multiply by 2.

@ryanmelt can you shed some additional light here?

GTI-Dakota, 2023-04-13:
I'll do the # of interfaces * 2 rule and work off of that. Thanks!

closed 2023-04-13 by GTI-Dakota