OpenC3/cosmos #594: tcpip_server_interface.rb does not call connect_reset and disconnect_reset on connect and disconnect, respectively.

GTI-Dakota, 2023-04-07:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Describe the bug**
In title

**To Reproduce**
Create a TCPIP Server with the connect_reset and disconnect_reset methods. Try printing to the Log with them.

**Expected behavior**
Nothing happens

Code references:
https://github.com/OpenC3/cosmos/blob/8eb694e84bbda5e5ca4c9520a05e3b375394ac7a/openc3/lib/openc3/interfaces/tcpip_server_interface.rb#L125

https://github.com/OpenC3/cosmos/blob/8eb694e84bbda5e5ca4c9520a05e3b375394ac7a/openc3/lib/openc3/interfaces/tcpip_server_interface.rb#L170

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
super() is not called, nor is connect_reset/disconnect_reset for each protocol, as is in:
https://github.com/OpenC3/cosmos/blob/8eb694e84bbda5e5ca4c9520a05e3b375394ac7a/openc3/lib/openc3/interfaces/interface.rb#L184
https://github.com/OpenC3/cosmos/blob/8eb694e84bbda5e5ca4c9520a05e3b375394ac7a/openc3/lib/openc3/interfaces/interface.rb#L196
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++

**Environment (please complete the following information):**
Standard

- OS: Any
- OpenC3 COSMOS Version 5.5.0, probably others
- Browser
  -- [x] Internet Explorer (IE)
  -- [x] Google Chrome
  -- [x] Chromium based browsers
  -- [x] Mozilla Firefox
  -- [x] Safari

ryanmelt, 2023-04-07:
TcpipServerInterface creates a StreamInterface for every connection.   connect/disconnect are called on these internal interfaces, which do call the connect_reset/disconnect_reset methods.   Protocols aren't used at the top level TcpipServerInterface, they are only attached to the internal StreamInterfaces.   See listen_thread_body lines 393 to 405 and the use of @protocol_info

GTI-Dakota, 2023-04-07:
Okay, I understand now. Thanks! my apologies for the misunderstanding

closed 2023-04-07 by GTI-Dakota