OpenC3/cosmos #597: Is there a way to send_raw through a router?

GTI-Dakota, 2023-04-11:
Hello,

I have an interface with two routers, and I need to send certain data through one router but not the other (without inect_tlm, if possible.)

I have tried the following code:

> router = get_router("MY_DATA_ROUTER") <- This works
> send_raw(router["name"], "\xFF") <- this does not 

The error I get is: `RuntimeError : Interface 'MY_DATA_ROUTER' does not exist`
This is not surprising. However, I need a protocol from INTERFACE A to send out ROUTER B. Is this possible without doing some weird `inject_tlm` shenanigans?

Thanks,

Dakota

GTI-Dakota, 2023-04-11:
Since I am in a protocol, is there some way to use the hash info to regenerate/attach to the object that a router microservice owns so that I can use its socket? I am open to any kind of manipulation even if slightly unsafe

GTI-Dakota, 2023-04-11:
This is the only way I figured out how to send it.... still sends through both routers but I could probably just add an IgnorePacketProtocol for that specific data:
`Topic.write_topic("#{$openc3_scope}__TELEMETRY__{#{packet.target_name}}__MY_DATA", { 'target_name' => packet.target_name, 'packet_name' => "MY_DATA", 'buffer' => packet.buffer(false) }, '*', 100)`

GTI-Dakota, 2023-04-11:
Either way, it creates an infinite loop. Not ideal

GTI-Dakota, 2023-04-11:
I'm just going to implement an interface to wrap the tcp_client_interface that the router uses so I can send the data via an interface_cmd

jmthomas, 2023-04-12:
Seems like you could apply the ignore packet protocol to filter out what you don't want:

```
    ROUTER MY_DATA_ROUTER ...
      MAP_TARGET DATA
      PROTOCOL READ IgnorePacketProtocol INST IMAGE # Drop all INST IMAGE packets
```

Also, we have both `interface_cmd` to send directly to an interface and `protocol_cmd` to send to one of the attached protocols.

GTI-Dakota, 2023-04-12:
I used the interface command and unfortunately it caused a stack overflow (Since it was being dispatched from a protocol, might be on me though). 

I have been unsuccessful with the IgnorePacketProtocol on both my routers, as it was unsuccessful on both routers I am using, so I've created a strange loopback. Not sure why this is happening

GTI-Dakota, 2023-04-12:
This code is dispatched when my interface detects a command called "CREATE_CSV", for example:
> def send_to_router(packet, router_name)
>     command = MY_ARGS.dup
>     command['route'] = 'create_csv'
>     packet.buffer = command.to_json
>     RouterTopic.router_cmd(router_name, 'create_csv', *command, $openc3_scope)
> end

I get my stack overflow here: `RouterTopic.router_cmd(router_name, 'create_csv', *command, $openc3_scope)`

GTI-Dakota, 2023-04-12:
In script runner, router command: `router_cmd('MY_DATA_ROUTER', 'test_command', 'test')`

> CommandError : ERR unknown command 'router_cmd', with args beginning with: 'MY_DATA_ROUTER' 'test_command' 'test' 'scope' 'DEFAULT'

Am I missing an import? I get the same error inside of a protocol as well

GTI-Dakota, 2023-04-12:
This same error is seen with the router_protocol_cmd, too

jmthomas, 2023-04-13:
`router_cmd` and `router_protocol_cmd` are both part of openc3/api/router_api.rb. If you do `include Api` inside your interface or router you'll get all of it.

GTI-Dakota, 2023-04-13:
[image]
This is what is currently in my protocol (that isn't able to send the command.) Is the import incorrect?

In Script Runner, include Api did not work, but include OpenC3::Api did, but then I got the following error:
[image]
[image]

jmthomas, 2023-04-13:
Script Runner should never `include OpenC3::Api` because it needs to use the scripting APIs. Your protocol should only do the include inside the class:
```
module OpenC3
class MyProtocol < Protocol
  include Api 
end
end
```

closed 2023-04-30 by ryanmelt