OpenC3/cosmos #600: Is it possible to generate a packet via a router?

GTI-Dakota, 2023-04-12:
Hello,

No matter what I try, I cannot generate a packet via a read_packet() method in a protocol that is tied to my router.

Here is some code I've used:
```
parsed_buffer = JSON.parse(packet.buffer)
if parsed_buffer.keys.include? "message"
    Logger.info("Log data found: #{parsed_buffer}")
    packet.target_name = "<%= my_target_name %>"
    packet.packet_name = "LOG_DATA"
    packet.accessor = OpenC3::JsonAccessor
    packet.received_time = Time.now.sys
    Logger.info(packet.to_json)
    return packet
end
```
I am trying to update telemetry in my system before the packet goes further than the router. I initially used "inject_tlm" but when I do that the packet is immediately re-sent back through the router as a loopback (due to it being new telemetry), as well as my OTHER router which is also tied to the same target.

(Related to this issue: https://github.com/OpenC3/cosmos/issues/597)

Is this possible or does a router not allow valid packets to be added to telemetry via the return packet?

Thanks,

Dakota

jmthomas, 2023-04-13:
The protocol `read_packet` method can't return additional packets but only modify the current one. The protocol `read_data` method _can_ return additional data which will get turned into additional packets. The initial call to `read_data` sends an empty string as data. This is the point where you can send extra data to create a new packet. The [burst_protocol.rb](https://github.com/OpenC3/cosmos/blob/main/openc3/lib/openc3/interfaces/protocols/burst_protocol.rb#L58) has some more details on this.

Something like (untested):
```
module OpenC3
  class MyProtocol < Protocol
    def initialize(allow_empty_data = nil))
      super(allow_empty_data)
      @buffer = []
    end
    
    def read_data(data)
      if (data.length <= 0)
        if @buffer.length > 0
          return @buffer.pop
        else
          return super(data)
        end
      else
        parsed_buffer = JSON.parse(data)
        if parsed_buffer.keys.include? "message"
            Logger.info("Log data found: #{parsed_buffer}")
            # Create a raw buffer of JSON that can be identified
            json = JSON.generate(???)
            @buffer.push(json)
        end
        return data
      end
    end
  end
end
```

GTI-Dakota, 2023-04-13:
Whoops, thanks for the info! I've got the packets working now, I think I got too far in the weeds yesterday and needed to take a step back.

closed 2023-04-13 by GTI-Dakota