OpenC3/cosmos #609: General flow from Router -> Interface -> Hardware

GTI-Dakota, 2023-04-14:
I was taking a look at your flowgraphs here: https://openc3.com/docs/v5/protocols and they are very helpful. 

After reading through your github code, as I understand it, if I have the following architecture: 

Client <-> Router <-> Interface <-> Hardware

When the router is reading/writing, it is not being passed data correctly, but rather through a Topic, which would mean that the data is passed to Redis in between the router and interface.

So in my head, the flow is:

Client <----TCP---> Router <----Redis Topic---> Interface <----TCP---> Hardware

Is this correct?

Thanks,

Dakota

GTI-Dakota, 2023-04-14:
Updated to include Redis in flow

ryanmelt, 2023-04-14:
That is correct.

closed 2023-04-14 by ryanmelt

GTI-Dakota, 2023-04-18:
> it is not being passed data correctly

I just realized that the "correctly" is supposed to say "directly". Totally changes the meaning of my sentence. My apologies