OpenC3/cosmos #619: Podman on Mac doesn't like the Z drive

scandey, 2023-04-18:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Describe the bug**
For whatever reason, Mac OS + Podman does not like the :Z drive used in the docker-compose. 

I found the secret here:
<https://github.com/containers/podman-compose/issues/509>

After manually replacing all ":z" in the compose file with ":x", Podman was able to finish building the containers and launch them. The Init container doesn't finish starting up though, so I'll have more bug reports related to this endeavor soon.

**To Reproduce**

Run cosmos-project on Mac using Podman, get 

`Error response from daemon: lsetxattr /Users/scott/HERMES/cosmos-testproject/cacert.pem: operation not supported`

**Expected behavior**

Podman correctly composes and launches the COSMOS containers.

**Screenshots**

**Environment (please complete the following information):**

- OS: 13.3.1 (22E261)
- OpenC3 COSMOS Version [e.g. 5.0.6]: 5.5.2
- Browser
  -- [] Internet Explorer (IE)
  -- [] Google Chrome
  -- [] Chromium based browsers
  -- [] Mozilla Firefox
  -- [x] Safari

ryanmelt, 2023-04-18:
Does :z work after properly configuring the user_ids?

ryanmelt, 2023-04-20:
I've confirmed that it doesn't like :z.    Removing the :z's fix it.  I'll add a special note to our Podman directions for Mac.
Thanks!

closed 2023-04-20 by ryanmelt