OpenC3/cosmos #621: What's the best way to send some data through the READ port (the read connection) of a TCP client?

GTI-Dakota, 2023-04-18:
Hello,

is there a way/what is the best way to send data through the read stream of an interface? I am creating an interface that send CMDs to port 8000 and receives tlm through 8001, but in order to "register" my cmd and tlm connections I must send a key to both endpoints. This is how the architecture is set up, as the endpoints are not owned by myself, but rather another application. My current plan is to create a custom interface that would send ONLY a specific command through the read port. Is this a good approach or is what I am describing architecturally impossible?

Thanks,

Dakota

ryanmelt, 2023-04-18:
You can do with with no to very little code with two separate interfaces.   One for commands to the target, and another for telemetry from the target.  

MAP_TLM_TARGET to one interface
MAP_CMD_TARGET to the other.

You might still want to do simple custom interfaces derived from TcpipClientInterface that override the connect method to call super() and also to send the necessary "register" commands.

GTI-Dakota, 2023-04-18:
Perfect. I think that will work. Thanks!

closed 2023-04-18 by GTI-Dakota