OpenC3/cosmos #623: Having trouble sending interface_protocol command

GTI-Dakota, 2023-04-20:
Hello,

I cannot send a protocol command via script runner. I have an interface with state CONNECTED, with a protocol that has the following method:

> def protocol_cmd(cmd_name, *cmd_args)
>             # do thing
>             Logger.info("I did thing: #{cmd_name}, with #{cmd_args}")
>             return true
> end

when attempting to call the interface_protocol_cmd in script runner:

`interface_protocol_cmd("MY_DATA_INT", "UPDATE_PARAMS", "PARAM_1", "PARAM_2")`

I get the following error: 

`2023/04/20 18:46:44.508 (2023_04_20_12_42_41_235_temp.rb:1): CommandError : ERR unknown command 'protocol_cmd', with args beginning with: 'MY_DATA_INT' 'UPDATE_PARAMS' 'PARAM_1' 'PARAM_2' 'read_write' 'READ_WRITE' 'index' '-1' 'scope' 'DEFAULT'`

I have tried all sorts of methods with no success. Can you give me an example of a successful interface_protocol_cmd?

Thanks,

Dakota

ryanmelt, 2023-04-20:
Are you on 5.6.0?

GTI-Dakota, 2023-04-20:
5.5.0. I have managed to get around the error by writing directly to the topic for now

ryanmelt, 2023-04-20:
interace_cmd and protcol_cmd are not going to work before 5.6.0.   Please let me know if you try again on 5.6.0

closed 2023-04-20 by ryanmelt

GTI-Dakota, 2023-04-20:
Right, looks to be that way since the interface calls the protocol command incorrectly on 5.5.0.