OpenC3/cosmos #625: Cannot uninstall Previously installed Gems after 5.6.0 update.

GTI-Dakota, 2023-04-20:
Hello, 

I have migrated COSMOS 5.5.0 -> 5.6.0, and none of my gems can be initialized because `:log, :log_raw` no longer exist, so the model crashes. Additionally, I cannot uninstall them to re-install them, because `:log, :log_raw` no longer exist.

From initialize:

> Stderr:
> /usr/lib/ruby/gems/3.1.0/gems/openc3-5.6.0/lib/openc3/models/interface_model.rb:94:in `initialize': unknown keywords: :log, :log_raw (ArgumentError)
> 	from /usr/lib/ruby/gems/3.1.0/gems/openc3-5.6.0/lib/openc3/models/model.rb:91:in `new'
> 	from /usr/lib/ruby/gems/3.1.0/gems/openc3-5.6.0/lib/openc3/models/model.rb:91:in `from_json'
> 	from /usr/lib/ruby/gems/3.1.0/gems/openc3-5.6.0/lib/openc3/models/model.rb:99:in `get_model'
> 	from interface_microservice.rb:361:in `initialize'
> 	from /usr/lib/ruby/gems/3.1.0/gems/openc3-5.6.0/lib/openc3/microservices/microservice.rb:51:in `new'
> 	from /usr/lib/ruby/gems/3.1.0/gems/openc3-5.6.0/lib/openc3/microservices/microservice.rb:51:in `run'
> 	from interface_microservice.rb:691:in `<main>'

From Uninstall:
Stderr:

> Error uninstalling plugin: DEFAULT: [gem_name].gem__20230417183739: ArgumentError : unknown keywords: :log, :log_raw
> /usr/lib/ruby/gems/3.1.0/gems/openc3-5.6.0/lib/openc3/models/interface_model.rb:94:in `initialize'
> /usr/lib/ruby/gems/3.1.0/gems/openc3-5.6.0/lib/openc3/models/model.rb:91:in `new'
> /usr/lib/ruby/gems/3.1.0/gems/openc3-5.6.0/lib/openc3/models/model.rb:91:in `from_json'
> /usr/lib/ruby/gems/3.1.0/gems/openc3-5.6.0/lib/openc3/models/model.rb:108:in `block in get_all_models'
> /usr/lib/ruby/gems/3.1.0/gems/openc3-5.6.0/lib/openc3/models/model.rb:108:in `each'
> /usr/lib/ruby/gems/3.1.0/gems/openc3-5.6.0/lib/openc3/models/model.rb:108:in `get_all_models'
> /usr/lib/ruby/gems/3.1.0/gems/openc3-5.6.0/lib/openc3/models/model.rb:116:in `find_all_by_plugin'
> /usr/lib/ruby/gems/3.1.0/gems/openc3-5.6.0/lib/openc3/models/plugin_model.rb:292:in `block in undeploy'
> /usr/lib/ruby/gems/3.1.0/gems/openc3-5.6.0/lib/openc3/models/plugin_model.rb:291:in `each'
> /usr/lib/ruby/gems/3.1.0/gems/openc3-5.6.0/lib/openc3/models/plugin_model.rb:291:in `undeploy'
> /usr/lib/ruby/gems/3.1.0/gems/openc3-5.6.0/lib/openc3/models/model.rb:171:in `destroy'
> /openc3/bin/openc3cli:462:in `unload_plugin'
> /openc3/bin/openc3cli:605:in `<main>'

ryanmelt, 2023-04-20:
:( There was supposed to be a migration that runs to handle this.  @jmthomas ?

GTI-Dakota, 2023-04-20:
No worries, just giving you a herads-up. I rolled back to 5.5.0, uninstalled all gems, updated to 5.60, and then re-installed all of them. Seems to be working now

jmthomas, 2023-04-21:
We're probably past this step but you should've seen `Checking for needed migrations...` in the cosmos-init log. Then something like `Running Migration: 20230319154100_log_stream` and `Running Migration: 20230413101100_remove_log.rb`. I'll see if I can reproduce.

ryanmelt, 2023-04-21:
Thanks Dakota.  We'll be doing a quick 5.6.1 release to address this.

ryanmelt, 2023-04-21:
closed by #626

closed 2023-04-21 by ryanmelt