OpenC3/cosmos #637: ERROR:  While executing gem ... (Errno::EACCES)

DavidWathen, 2023-04-24:
A colleague is getting this error when creating a plugin. 

ERROR:  While executing gem ... (Errno::EACCES)
    Permission denied @ rb_sysopen - openc3-draco-cdh-1.0.7.20230421001456.gem

Any ideas on what could be causing this. 

Thanks,

David

delaney9693, 2023-04-28:
Hi! I've recently run into the same issue in Ubuntu 22.04 when trying to build the .gem. 

Initially, I was running into the same issue when building/running as a user in the sudo group, and the ownership within the container was mostly by root. I did a fresh install of Docker, and a new clone of the cosmos repo by my non-sudo user, then built and deployed the containers as the non-sudo user. 

From what I can see (specifically in the cmd-tlm-api container, in the /gems/ directory), ownership of most subdirectories and files isn't by root or openc3, but the user id number and group number of the non-root user on my host machine that built and launched the containers. 

I'm relatively new to Docker/Docker Compose, so I apologize if this is in any way me misusing anything (I didn't change anything in the .yaml file that I can think of...), but I'm pretty sure I followed the OpenC3 install instructions in their entirety. 

Thanks!

ryanmelt, 2023-04-29:
On our newer releases, the userid/groupid should be the same as the user on the host machine.  This allows for mounting host volumes and reading/writing successfully from the containers.

closed 2023-05-19 by jmthomas