OpenC3/cosmos #639: Logging level "debug" is not defined for Logger:Class

GTI-Dakota, 2023-04-24:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Describe the bug**
Out-of-the-box Logger that can be referenced in an OpenC3 protocol file cannot call the method "debug"

**To Reproduce**
Create a protocol, attach to interface, attempt to call `Logger.debug("MY DEBUG MESSAGE")` in protocol.

**Expected behavior**
` Unable to require my_protocol.rb due to undefined method 'debug' for Logger:Class (NoMethodError)`

**Screenshots**
[image]

**Environment (please complete the following information):**

- OS: N/A
- OpenC3 COSMOS Version: 5.6.0
- Browser
  -- [x] Internet Explorer (IE)
  -- [x] Google Chrome
  -- [x] Chromium based browsers
  -- [x] Mozilla Firefox
  -- [x] Safari

ryanmelt, 2023-04-24:
Is your protocol in the OpenC3 namespace? ie. module OpenC3?
If not you'll need to do OpenC3::Logger

GTI-Dakota, 2023-04-24:
I tried the OpenC3::Logger and it did not work. The file uses namespace OpenC3. Is the `include OpenC3::Api` screwing things up? That's been in there for 4 months, but I believe that it was required.

[image]

ryanmelt, 2023-04-24:
I don't recommend including Api at the top level, that line should be inside your class.  But probably not the problem.

Our Logger class definitely defines .debug: https://github.com/OpenC3/cosmos/blob/main/openc3/lib/openc3/utilities/logger.rb#L123

There is something else going on.  What files are you requiring?

GTI-Dakota, 2023-04-24:
[image]

Here's what I've got. This is only 1 protocol; I haven't gotten debug messages to work in various protocols across other gems either.

I'm pretty confused as well because debug is definitely defined in the source. Probably my bad but the solution is not immediately obvious to me. Is it possible that the Logger object is actually not the OpenC3:Logger somehow and is referencing something else?

ryanmelt, 2023-04-24:
Anything is possible.   We ship protocols that use the OpenC3::Logger (though not debug level).   
I would dig into the two redacted files (or eliminate them and see if it starts working)

GTI-Dakota, 2023-04-24:
Will do. Thanks! I will update you with the result.

ryanmelt, 2023-05-10:
Closing as I don't believe this is a COSMOS bug

closed 2023-05-10 by ryanmelt