OpenC3/cosmos #645: Is there a way to save a minio file to local machine via a script?

GTI-Dakota, 2023-05-01:
Hello,

Is there a way to grab a file in the script runner and then save it to a local machine like the data extractor does? I noticed that COSMOS 5 deprecated the save_file_dialog, so I was wondering if there is another way to do this. The flow would be as follows:

- Script starts
- Create File
- Save file to minio
- Open Save File Dialog so user can save to local machine
- Script Ends

Thanks,

Dakota

jmthomas, 2023-05-01:
You can use [put_target_file](https://openc3.com/docs/v5/scripting#put_target_file) to write to Buckets.  Also see [target_file.rb](https://github.com/OpenC3/cosmos/blob/main/openc3-cosmos-init/plugins/packages/openc3-cosmos-demo/targets/INST/procedures/target_file.rb) from the DEMO.

GTI-Dakota, 2023-05-01:
Well, I'm not looking to save to minio, I'm looking to pull from minio to save to my machine via a script

GTI-Dakota, 2023-05-01:
Is there a way to leverage the same API that the data extractor uses to allow a user to save a file from minio? Like in the bucket explorer with the download button. Is that capability able to be called via script?

jmthomas, 2023-05-01:
We have the `open_file_dialog` and `open_files_dialog` but don't yet have the `save_file_dialog` method. We can add this in an upcoming release. Right now the work around would be to tell the user that you saved it to a specific location and then open Bucket Explorer to that location so they can download it.

GTI-Dakota, 2023-05-02:
>Right now the work around would be to tell the user that you saved it to a specific location and then open Bucket Explorer to that location so they can download it.

We do that now! It is pretty helpful as is but we want that final leap to get the file to our computer after a massive suite has been ran, so we can easily save it to our shared drives. Thanks for adding it in the next release!

closed 2023-05-06 by jmthomas

ryanmelt, 2025-09-19:
Its a bug.  Still working in Ruby but looks like Python is broken.

timothyscherer, 2025-09-19:
Should I submit a new issue or is it covered? Any chance a fix for this works its way into 6.8.2?

ryanmelt, 2025-09-22:
I created an issue.  We'll get it fixed for 6.9.0 this week.

PR #648 (2023-05-06): Add download_file API
closes #645