OpenC3/cosmos #646: What is the most efficient way to maximize data passthrough throughput?

GTI-Dakota, 2023-05-01:
Hello, 

This issue is related to: https://github.com/OpenC3/cosmos/issues/609

I am looking to speed up my data throughput from Client <-> Router <-> Interface <-> endpoint

If possible, I would like bidirectional data flow as fast as possible through the signal chain. After previous issues created and some lessons learned, my current architecture is as follows:

CLIENT <-bidirectional-> DATA_ROUTER---cmd-> DATA_CMD_INT   -----> cmd_endpoint
_________________________________________^<---tlm---DATA_TLM_INT    <----- tlm_endpoint
                                                                
Would it be efficient to do an @interface_write(data) from my router to expedite the flow of data from router -> interface?
Would it be efficient to tweak the TLM_BUFFER_DEPTH? (depth to say, 10?)

Things I have already done:
- Set Docker Operator Container as priviledged
- Upped the Workers in the mounted puma.rb file
- Changed TLM_BUFFER_DEPTH to 200 (very slow TLM response)
- Changed TLM_BUFFER_DEPTH to 10 (definitive speed improvement, no `Log writer out of order time detected` messages generated during high traffic time)
- Tried to reduce operational lag in protocols that are attached to the cmd interface and tlm interfaces

This architecture is designed to be a highway, and I want to give myself the best shot at achieving that.

Thanks,

Dakota

ryanmelt, 2023-05-02:
Want to do a call today Dakota?

GTI-Dakota, 2023-05-02:
Thanks for the help @ryanmelt

closed 2023-05-02 by GTI-Dakota