OpenC3/cosmos #647: Scheduling a group of events by file to the calender

bradenvirdellgenevatech, 2023-05-02:
Hello,

We are looking for a way to create multiple "timeline activities" to the calendar through uploading of a schedule instead of inputting them all individually. Where we can set what procedure is ran, at what time, and any variables/metadata that we need to send to it. It seems like all the functionality is there except for a schedule upload capability. Would this be something that could be added or is possibly already available?

Thanks,
Braden

ryanmelt, 2023-05-10:
This ticket has a lot of thumbs up. 
What do you actually want though?

1. A ruby client for accessing the Calendar API?
2. A specified calendar file format to upload?
3. Something like table manager upload/download where it could somehow handle an arbitrary schedule format?

bradenvirdellgenevatech, 2023-05-10:
1. If there is a possibility of API Access where we can talk to the calendar through the script runner I think that would be great and the most advantageous, at least for my situation.
2. I have attached what I believe would be a good file format that could be uploaded to the calendar, only issue would be if someone has a LOT of environment variables. 

I think these 2 would be a good combo for those who don't want to use the API they can just upload and for those who would like more control they would have more access.

Hope this helps.

[calendar_upload.txt](https://github.com/OpenC3/cosmos/files/11443062/calendar_upload.txt)

MTI-twalker, 2023-05-10:
Also, this may be something that can be covered by having a similar function as outlooks "Recurrence" form:
[image]

jmthomas, 2023-05-10:
We actually already have a ticket #3 for recurring events in Calendar.

jmthomas, 2025-07-22:
We now have recurring events and a [Schedule](https://github.com/OpenC3/openc3-schedule) framework where this would be implemented.

closed 2025-07-22 by jmthomas