OpenC3/cosmos #650: Command META field not working as expected

finchlt, 2023-05-03:
I'm trying to use the data from the META field in a command from a protocol.

For each command for the target a set the following metadata. The actual value changes from command to command. 

```
META RECV_LEN "7"
```
I then try to access this metadata from the `write_packet` method of a protocol using the snippet below.

```ruby
      packet_metadata = packet.meta()
      Logger.info("Packet meta #{packet_metadata.to_s}")
      if packet_metadata.has_key?("RECV_LEN")
        return_data_cnt = [packet_metadata["RECV_LEN"].to_i]
      else
        Logger.info("This isn't working")
        return_data_cnt = [0]
      end
```

Then when i send the command with the META field set its logged that the current packet has an empty META data hash?

Am i using this right or is there an issue with META fields not being persisted?

ryanmelt, 2023-05-03:
They should be available for the common cases.   

There are cases where they wouldn't be, such as if the command was unidentified, or an earlier protocol modified the packet.

We will try to recreate.

finchlt, 2023-05-03:
I'm only running the single protocol. Also let me know if you need more information as well to recreate.

ryanmelt, 2023-05-09:
Just tried to recreate this and it worked fine.   

Is your META line below a command parameter?   If so it will get associated with the parameter and not the overall packet.

Also, a note on your example code: metadata_values are stored in an array, because it is a key to multiple values based system.

For example, I added META DUDE SWEET to a packet, and your Logger statement printed out:
```Packet meta {"DUDE"=>["SWEET"]}```

Note that "SWEET" is in an array.  So you will need to dereference the first array element before calling .to_i

closed 2023-05-19 by jmthomas