OpenC3/cosmos #662: PLUGIN.txt diff tool during plugin upgrade does not insert final line correctly

finchlt, 2023-05-12:
**Describe the bug**
When using the diff tool for PLUGIN.txt when upgrading a plugin if the line added by the new PLUGIN.txt is the final line in the file before and after, the diff tool appends the new line to last line of the old PLUGIN.txt file resulting in an incorrect file by default.

**To Reproduce**
1. Create valid original plugin.txt with no empty line at end of file. 
2. Load original plugin
3. Update plugin.txt and plugin gem to include one additional line at end of plugin.txt (again with no empty line at end of file)
4. Load new plugin gem through Admin panel, navigate to diff at and of file, press diff insert button. 
5. See that diff from right side is added to the final line on left side instead of as new line (this is invalid and easy to do if you just wanted to quickly accept all diffs)

**Expected behavior**
Add diff from right side as newline on left side since this reflect what the diff is intended to be. 

- OS: Ubuntu 20.04
- OpenC3 COSMOS Version 5.6.1
- Browser
  -- Firefox

jmthomas, 2023-05-13:
Thanks. This diff is new since [5.6.0](https://github.com/OpenC3/cosmos/releases/tag/v5.6.0) so we'll fix this in a future update.

closed 2023-05-22 by jmthomas

jurgensymynck, 2023-07-27:
Hi, 

I assume issue #760 is linked to this bug, so might not be solved yet.

gr. Jurgen

PR #672 (2023-05-22): Ace diff
closes #662