OpenC3/cosmos #665: Is there a way to generate a TLM Definition given a hash?

GTI-Dakota, 2023-05-16:
Hello,

I have a JSON string retrieved from hardware that I convert to a hash. Is there an easy way to "create" a TLM def if I save off that string/hash? Or is this something that I will just need to write a quick script for? Additionally, it looks like TLM defs do not natively support boolean types. Am I correct?

Thanks,

Dakota

jmthomas, 2023-05-16:
I think you'll have to create a script if you want to automate generating a telemetry definition.

We basically support Boolean types by providing states such as:

```
    APPEND_ITEM BOOL 8 UINT "Boolean"
      STATE FALSE 0
      STATE TRUE 1
```

GTI-Dakota, 2023-05-16:
Sounds good. Thanks for the info!

closed 2023-05-19 by jmthomas