OpenC3/cosmos #667: Missing Data From 1 Target in a Particular Time Frame

GTI-Cody, 2023-05-17:
I'm missing data from only 1 target during a specific period of time. That time period was during a test which went well, but that I need to reference data from. Checks utilizing that target passed at that time so we were getting data, it just appears to not have been saved. I have data for all of our other targets before during and after that time period. 

Is there anything else I could check that could get the needed data?

GTI-Cody, 2023-05-17:
From a quick look through of the tlm binary in our log bucket, it looks like the data is missing there too. I searched for the first occurrence of a packet from that target and it has the same time as the data we extracted.

ryanmelt, 2023-05-17:
Is there a log file for the target in the bucket?   If so, everything COSMOS received should be logged in it.

GTI-Cody, 2023-05-17:
There is a log file in the bucket. 

That doesn't make too much sense as we managed to run through an entire test suite in that time where COSMOS didn't receive any data (according to the log file and our extracted data). The Script Runner logs and our manual logging agree with the time that the test occurred. As well as data from other targets showing valid data for that test. This one target just happens to have a hole in the data there. 
Just to clarify, if we weren't receiving data from this target then we wouldn't have been able to test anything.

ryanmelt, 2023-05-17:
How big is the hole?

The Current Value Table would still have old data in it, even if data wasn't flowing for a period.   Did your test validate values changing?

GTI-Cody, 2023-05-17:
~10 minutes of data is missing.

The test validated values changing.

ryanmelt, 2023-05-17:
10 minutes is how long each log file is... Is there a complete file missing?  ie. file timestamps show an entire file missing?

ryanmelt, 2023-05-17:
If the system is still up, can you dump the operator container docker logs?

GTI-Cody, 2023-05-17:
There are no earlier files in the day for this target. This test was first thing in the morning, and we only get data on this target when the device under test is powered on.

GTI-Cody, 2023-05-17:
We will send the Docker logfile your way.

bradenvirdellgenevatech, 2023-05-18:
We ran into this issue again today. It appears that the first test of the day will have lost data. After ~10 minutes from powering on our device under test we start getting data for it.

PR #670 (2023-05-23): Update BufferedPacketLogWriter to empty buffer when closing file
closes #667
closed 2023-05-23 by ryanmelt