OpenC3/cosmos #669: Accessing File from outside the microservices folder with a microservice

bradenvirdellgenevatech, 2023-05-18:
Hello,
We were wondering if it is possible to require/load a script in a microservice that doesn't exist in the microservices folder? 

This is more or less my file structure: 
├── targets/TEST_TARGET
│....└── lib
│...............├── script.rb
│...............└── other useful scripts
└── microservices/MICROSERVICE
.........└── microservice.rb

I need to load script.rb in microservice.rb on startup, and script.rb loads lots of other ruby files so I can't just move the one script into microservices. Trying `load "TEST_TARGET/lib/script.rb"` and variants thereof have been unsuccessful. Is there any way to accomplish this?

Thanks!

ryanmelt, 2023-05-18:
Its possible, but the code to do that is currently all in ScriptRunner.   Can you move the script.rb file to a top-level lib folder?   If you do that, it will "just work".

bmartinGTI, 2023-05-18:
I was able to require the script.rb from the top level lib folder, but I'm not able to require other scripts in the targets/TEST_TARGET/lib folder in script.rb. This file also needs access to files in other gems (typically it's been run in script runner but I'm trying to adapt it to be used by a microservice). Is there a specific syntax for requiring these other files from the top level lib folder, or is it something that's not accessible/in another docker container?

jmthomas, 2023-05-19:
So now you have something like this with script.rb in the top level lib:
```
|- lib
│-- script.rb
|- targets
|-- TEST_TARGET
|--- lib
|---- <other useful scripts>
│- microservices
|-- MICROSERVICE
|--- microservice.rb
```
You want to call script.rb from microservice.rb and you want script.rb to call various files from TEST_TARGET/lib? Is that correct?

bmartinGTI, 2023-05-19:
Exactly, yes.

jmthomas, 2023-05-22:
I can reproduce the error you're talking about. We're working on it ...

ryanmelt, 2023-05-23:
PR will let other places work just like ScriptRunner. 
You will need to add require 'openc3/utilities/bucket_require' to your script before requiring something that pulls a file from a bucket.

closed 2023-05-23 by ryanmelt