OpenC3/cosmos #671: tcpip_server_interface question

rmslick, 2023-05-18:
Beyond adding a port entry into the openc3-operator service in compose.yaml and specifying tcpip_server_interface in the interface - is there anything required to send data from a remote machine to the host machine and into the cosmos container? 

All I have done is specified I have tried this both on Windows 10 and RHEL8 and I see no data appearing in cosmos in either configuration. 

INTERFACE <%= my_target_name %>_INT tcpip_client_interface.rb host.docker.internal 8080 8081 10.0 nil BURST
  MAP_TARGET <%= bob_target_name %>

compose.yaml:

ports:
   # Tried both of the following
   #8080:8080 
  127.0.0.1:8080:8080

ryanmelt, 2023-05-18:
1. On the tcpip server machine, you'll need to NOT use 127.0.0.1 in compose.yaml to the operator container (just 8080:8080)
2. Your client interface should probably be 8080 8080, not 8080 8081
3. The only other thing is typically host firewall settings

rmslick, 2023-05-18:
1. So the note above was to imply that I tried both ways, 8080:8080/8181:8181 does not work either.
2. I don't have a client interface, I mistyped and meant tcpip_server_interface.rb, my bad
3. I have no idea how to diagnose if this is the case. My test set up involves a python tcp client running on the same host that runs the docker container. The python is sending to host IP '127.0.0.1' at the port specified in the interface. If you have any suggestions for troubleshooting, that would be appreciated. Thanks

ryanmelt, 2023-05-18:
Host firewall won't be a problem if everything is on the same machine.
Is the TCP/IP client connecting?

rmslick, 2023-05-18:
Yes it is connecting.And when I list what is listening to 8080 on RHEL I see cosmos has bound to it. I am not packets in the format specified by the telemetry in the plugins, but I assume that the Rx bytes would increment. Though prior to writing the test script, I was sending it properly formatted telemetry packets from the telem source and it also did not work

rmslick, 2023-05-18:
I was able to make it work. I was specifying the port in the wrong service: openc3-traefik and not openc3-operator. It is behaving as expected and the RX count is incrementing. Thank you

ryanmelt, 2023-05-18:
👍

closed 2023-05-18 by ryanmelt