OpenC3/cosmos #682: ScriptRunner can't get tlm of block

tj-scherer, 2023-05-25:
I am trying to run a script in the way described by the documentation [here](https://openc3.com/docs/v5/script-runner). My script contains an equivalent to the following, where `COLLECTS` is a block data type:
`puts tlm("INST HEALTH_STATUS COLLECTS")`

Is there another way to read out block data types? Just typing 'puts' results in a blank line, but the above command results in nothing at all in the debug log.

COSMOS v5.7.2 running on Ubuntu 22.04, using cosmos-project.

tj-scherer, 2023-05-25:
I also tried this with an array item, and see the same result. Works perfectly fine for UINT items.

tj-scherer, 2023-05-26:
Yes the data is there on packet view and the telemetry viewer.

If I add `.formatted` to the end of an ARRAY_ITEM, I do see the data, but there is still a random 'r' at the end, and not sure why an address is showing either:
`00000000: 0F FF 00 72 r`

I need to wait for some hardware to be reconnected to test the BLOCK type with .formatted, will update when I can.

jmthomas, 2023-05-26:
The `formatted` method is a COSMOS method that prints out the address, raw bytes, and ASCII representation. 0x72 is 'r' in ASCII. When you try to simply `puts` this data it doesn't work because it's not ASCII data that can be printed.

tj-scherer, 2023-05-26:
Seems likely that the block would work the same way then? This should be included in the Script Runner documentation if it is working as expected.

tj-scherer, 2023-05-26:
Ultimately this probably comes down to familiarity with ruby, the example was the primary thing I had used to attempt what I wanted, so that would be the best place for it.

The telemetry docs also say that "The only difference between a STRING and BLOCK is when COSMOS reads a STRING type it stops reading when it encounters a null byte (0)." which does seem a little misleading to me. I also wasn't aware of the option to append ".formatted" (I had only seen a second argument with type: :RAW, etc) to this type.

Overall, I think it might be best to have a separate section of the docs dedicated to data types, since they are shared across command, telemetry, screens, scripts, etc. Most things revolving around UINT have been extremely intuitive, but it would be nice to know how to handle ARRAY/BLOCK/STRING types across each of the different interfaces.

jmthomas, 2023-05-26:
Thanks for the feedback!

closed 2023-05-26 by jmthomas

jmthomas, 2023-05-30:
closed by https://github.com/OpenC3/openc3.com/pull/61