OpenC3/cosmos #698: Read_conversion variable save to json

MTI-GMays, 2023-06-13:
**Is your feature request related to a problem? Please describe.**
unable to output read conversion's passed function variable to json

**Describe the solution you'd like**
Add another variable like args_passed as a read attribute in conversion.rb to just a save the variable passed in to the function to json. 

[image]

jmthomas, 2024-09-02:
It looks like `accessor.rb` is storing the args similar to this suggestion. Add `@args` to conversion, edit packet_item and packet_item_spec.rb::self.from_hash to ensure conversions can "round trip".  May also have to modify packet_config.

Ensure changes across both Ruby and Python.

jmthomas, 2025-07-09:
closed by #802

closed 2025-07-09 by jmthomas