OpenC3/cosmos #699: Screen Errors, TLM Freezes

tsjordan-eng, 2023-06-16:
Sometimes I my telemetry viewer screen freezes, but I don't notice it (values don't turn pink). I have to click the little warning sign on the top if I notice it. 
[image]
 When I do I get a cryptic message like `1686876547300: (usage)`. I don't know what causes this. Is there a way to just reset all the errors on the screen or to track down this error?

jmthomas, 2023-06-16:
The code to display the error looks like this: `${error.time}: (${error.type}) ${error.message}`. So it's a time with the type as usage and then it's supposed to display the error message. Try clicking the Edit button which should display the errors with additional information: 
```
          let msg = `At ${error.lineNumber}: (${error.line}) ${error.message}.`
          if (error.usage) {
            msg += ` Usage: ${error.usage}`
          }
```

tsjordan-eng, 2023-06-19:
When I get the error, now I click the pencil to edit the screen (instead of the warning button). The only error I get is:
```
At undefined: (undefined) .
```

jmthomas, 2023-06-19:
I've got a feeling that `get_tlm_values` is throwing an error and we're handling it like other usage error types on line 734 in Openc3Screen.vue. If you see this happen try going to the console and see if there are any errors thrown in the cmd-tlm-api container:
```
docker logs cosmos-openc3-cosmos-cmd-tlm-api-1
```

tsjordan-eng, 2023-06-20:
Okay, I'll try this next time it happens. It's usually around when targets get disconnected then reconnected.

ryanmelt, 2023-06-21:
We've seen this a few times now.  Going to flag this as a bug. 
1. Need to provide better error information
2. Need to make a "dead" screen more obvious
3. Need to have screens recover if the api goes away for a bit (like restarting the cmd-tlm-api container), and then comes back

PR #720 (2023-06-30): Better handle screen tlm request timeouts
closes #699
closed 2023-06-30 by jmthomas