OpenC3/cosmos #703: Installing gems that need extra Libs to complete

bradenvirdellgenevatech, 2023-06-21:
Hello,

I have been using the openc3s ability to install 3rd party ruby gems through the .gemspec file. I am have good success with most of the gem except for "rmagick". From what I gather I am missing the lib "imagemagick" and "imagemagick-dev". What would be the best way to get these libs into the proper container (I assume it is the operator).
[image]

jmthomas, 2023-06-21:
It's been a while since I've played with rmagick but it's a pain in the butt due to all the dependencies. I think you'll have to build your own custom operator container and add the required packages.

closed 2023-06-21 by jmthomas

bradenvirdellgenevatech, 2023-06-21:
That's what I have been attempting, I will keep trying a few things.

jmthomas, 2023-06-21:
Here's a thread where someone got v6 to work: https://github.com/rmagick/rmagick/issues/834