OpenC3/cosmos #736: Odd Script runner behavior with suites and longer scripts

bradenvirdellgenevatech, 2023-07-05:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Describe the bug**
While running our suites and some of our longer scripts we see issues where the script runner instrumentation will pause/stop on a line but the script will continue on through the prompts. Most the time the instrumentation will start following correctly again after the line has completed, but not always, we have seen stops on, cmds, checks, wait_checks and elsif statements.

**To Reproduce**
We have no exact way of replicating this as it is happening during relatively long scripts. We are willing to show you the behavior over a video call.

**Expected behavior**
We expect instrumentation to follow through scripts without random instrumentation freezes and to be on the line it is actually running at the time. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

- OS: Ubuntu 22.04.2
- OpenC3 COSMOS Version 5.8.1
- Browser
  -- [x] Internet Explorer (IE)
  -- [x] Google Chrome
  -- [] Chromium based browsers
  -- [] Mozilla Firefox
  -- [] Safari

ryanmelt, 2023-07-05:
Anything like this should be repeatable.
If you goto Script -> Instrumented Script, and send us that for the misbehaving script that would be helpful.

bradenvirdellgenevatech, 2023-07-05:
I have emailed you due to the nature out the outputs, I would rather not have them in a public domain.

ryanmelt, 2023-07-05:
closed by #737

closed 2023-07-06 by ryanmelt