OpenC3/cosmos #741: Authentication error occurs when trying to create a StreamingWebSocketApi object in a microservice

bmartinGTI, 2023-07-10:
When running the line `OpenC3::StreamingWebSocketApi.new do |api|` in a function (save_data_to_csvs) in a microservice, the following authentication error is thrown:

```
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.8.1/lib/openc3/utilities/authentication.rb:164:in `_make_request': authentication request failed 
request uri: http://openc3-keycloak:8080/auth/realms/openc3/protocol/openid-connect/token 
header: {"Content-Type"=>"application/x-www-form-urlencoded", "User-Agent"=>"OpenC3KeycloakAuthorization / 5.8.1 (ruby/openc3/lib/utilities/authentication)"} 
body: username=service&password=servicesecret&client_id=api&grant_type=password&scope=openid ::: 
response status: 401 
header: {"referrer-policy"=>"no-referrer", "x-frame-options"=>"SAMEORIGIN", "strict-transport-security"=>"max-age=31536000; includeSubDomains", "cache-control"=>"no-store", "x-content-type-options"=>"nosniff", "pragma"=>"no-cache", "x-xss-protection"=>"1; mode=block", "content-type"=>"application/json", "content-length"=>"72"} 
body: {"error":"invalid_grant","error_description":"Invalid user credentials"} (OpenC3::OpenC3AuthenticationError)
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.8.1/lib/openc3/utilities/authentication.rb:118:in `_make_token'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.8.1/lib/openc3/utilities/authentication.rb:82:in `block in token'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.8.1/lib/openc3/utilities/authentication.rb:78:in `synchronize'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.8.1/lib/openc3/utilities/authentication.rb:78:in `token'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.8.1/lib/openc3/script/web_socket_api.rb:129:in `connect'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.8.1/lib/openc3/script/web_socket_api.rb:39:in `initialize'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.8.1/lib/openc3/script/web_socket_api.rb:179:in `initialize'
	from /usr/lib/ruby/gems/3.2.0/gems/openc3-5.8.1/lib/openc3/script/web_socket_api.rb:302:in `initialize'
	from COSMOS_UTIL/procedures/data_extraction.rb:137:in `new'
	from COSMOS_UTIL/procedures/data_extraction.rb:137:in `save_data_to_csvs'
	from COSMOS_UTIL/procedures/data_extraction.rb:37:in `block (2 levels) in save_targets_to_csvs'
```
This is my plugin.txt:
```
# Set VARIABLEs here to allow variation in your plugin
# See https://openc3.com/docs/v5/plugins for more information
VARIABLE ops_target_name OPS

# Modify this according to your actual target connection
# See https://openc3.com/docs/v5/interfaces for more information
TARGET OPS <%=ops_target_name %>

MICROSERVICE LIGHTS_OUT LIGHTS_OUT_MS
    WORK_DIR .
    CMD ruby lights_out_microservice.rb
    TARGET_NAME <%= ops_target_name %>
    ENV OPENC3_API_USER service
    ENV OPENC3_API_PASSWORD servicesecret
    ENV OPENC3_SERVICE_PASSWORD openc3service
```

The same code runs without issue via Script Runner. Is there a setting somewhere in the .env file or elsewhere that would enable this functionality in a microservice? Currently seeing this issue on version 5.8.1

ryanmelt, 2023-07-10:
Do you have an account in Keycloak with username service, and password servicesecret?

bmartinGTI, 2023-07-11:
I swapped OPENC3_API_USER and OPENC3_API_PASSWORD with some actual keycloak credentials, it appears to be working now 👍 thanks!