OpenC3/cosmos #744: Inject telemetry from protocol

finchlt, 2023-07-11:
I have a custom protocol that builds, and aggregated command buffer made up of many commands trigger by a command being sent. When this protocol runs for the given input command target and command I want to inject a tlm packet back with some metadata i can use for validation of telemetry generated by the aggregated command being executed. 

Is there an easy or best way to inject a TLM packet from a protocol. I tried a bunch of imports but couldn't get the `inject_tlm` function from script api to resolve.

ryanmelt, 2023-07-11:
inject_tlm can be used by
```
require 'openc3/api/api'
...
class MyProtocol < Protocol
  include Api

  def write_data(...)
    inject_tlm(...)
  end
end
```

You can't ever require 'openc3/script' from interfaces or protocols because those use HTTP to hit the actual API and require credentials.   You can use the API code directly though.

finchlt, 2023-07-12:
thanks! works perfect!

closed 2023-07-12 by finchlt

xavier-rangel, 2023-07-13:
> inject_tlm can be used by
>
> ```
> require 'openc3/api/api'
> ...
> class MyProtocol < Protocol
>   include Api
>
>   def write_data(...)
>     inject_tlm(...)
>   end
> end
> ```
>
> You can't ever require 'openc3/script' [...]

I have a custom protocol that is using 'openc3/script' for access to inject_tlm(), should I using 'openc3/api/api'?

ryanmelt, 2023-07-13:
@xavier-rangel Yes, don't use openc3/script (except for scripts).  Internal microservices should use openc3/api/api.