OpenC3/cosmos #745: Connect_Interface does not change config_params

bradenvirdellgenevatech, 2023-07-12:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Describe the bug**
Interface params do not update when connect_interface is used to reconnect an interface with different parameters than are given from the plugin when the gem is initially loaded

**To Reproduce**
Ran the code in the first screen shot

**Expected behavior**
I would expect the "config_params" to update when the "connect_interface" function is called.

**Screenshots**
[image]

[image]

**Environment (please complete the following information):**

- OS: Ubunutu 20.04
- OpenC3 COSMOS Version 5.9.0
- Browser
  -- [] Internet Explorer (IE)
  -- [X] Google Chrome
  -- [] Chromium based browsers
  -- [] Mozilla Firefox
  -- [] Safari

ryanmelt, 2023-07-17:
I'm torn on this.   get_interface() captures what was set in the plugin, and what will be used if you shut everything down, and then restart.    Modifying it will change that.   Maybe there should be a new API or a flag to indicate if you want the original or the current?

bradenvirdellgenevatech, 2023-07-17:
That would achieve what I am aiming for, as all I want to do is be able to confirm through a script my current configuration (ip, ports etc..)  after a change has happened as our system needs to be switched relatively often. Currently im tracking this with a tlm point but I feel like that can end up out of sync by accident. So the api/flag would work perfectly fine for my situation.

jmthomas, 2024-09-02:
When `connect_interface` is called it calls `InterfaceTopic.connect_interface` which is processed by the `interface_microservice` `run` method which calls `attempting(*params)`. That call recreates the interface and updates the `InterfaceStatusModel`. However, the `InterfaceStatusModel` does not put the `config_params` in the json and thus only the original params are returned by the `get_interface` due to the `InterfaceModel.get` call. 

If we add `config_params` to the `interface` `as_json` method it would make the `get_interface` api return the latest parameters. I think that should be the default behavior as it makes sense to return the latest values. We could add another parameter `original = false` which when true we ignore the `config_params` returned by `InterfaceStatusModel` and instead use the original values returned by `InterfaceModel.get`.

jmthomas, 2025-12-11:
closed by https://github.com/OpenC3/cosmos/pull/2267