OpenC3/cosmos #752: Log file locations of COSMOS tool gems (errors with Telemetry screens) ?

jurgensymynck, 2023-07-25:
While getting my 5.0.8-ish Telemetry screens working on the latest version of COSMOS (5.9.1) I run in an error message, which can be read when clicking the 'Errors' icon on the upper left corner of the faulty screen:

`1690274765803: (error) Error: Unknown widget: <%Widget`

My question: where can I find more context about this message? Where are the log files located of the parsing of the telemetry screens? (maybe i need to create another issue for the error itself, but this one is about the logs)
So far I tried the following: 
- A github search on the cosmos repo (https://github.com/search?q=repo%3AOpenC3%2Fcosmos+%22Unknown+widget%22&type=code) turns up `openc3-tool-common` and `openc3-cosmos-tool-dataviewer` as potential searches of this error message. 
  I tried to find a 'view log' function in the administration section of COSMOS -> Gems but that doesn't exists.
- in Docker, I tried to search for 'Unknown widget' in all container logs (first tab in the series Logs - Inspect - Terminal - Files - Stats), to no avail
- in Docker, I tried `grep -nrw -iF -s -e "Unknown widget" / `, to no avail

thanks in advance, Jurgen

- windows 10
- Google Chrome Versie 115.0.5790.102 (Officiële build) (64-bits)
- OpenC3 COSMOS 5.9.1
- Docker Desktop v4.21.1
- ruby 3.2.2 (2023-03-30 revision e51014f9c0) [x64-mingw-ucrt]

jurgensymynck, 2023-07-25:
Update: I also browsed through the COSMOS web interface -> Bucket explorer, but found nothing usable.

Bucket explorer -> Volumes: `/gems` -> `/Build Info/`

Al these .info files are useless; they contain only "--no-document", a build flag that is used in a lot of places (https://github.com/search?q=org%3AOpenC3%20%22--no-document%22&type=code) ...

gr. Jurgen

jmthomas, 2023-07-25:
If a widget can not be found then you get the Unknown widget error. Judging by the `<%` this is probably because you have an ERB formatting error. ERB open tags must be `<%` and closed by `%>`. Note that those types of tags are purely for logic control and do NOT insert anything into the file. You must use the `<%=` opener to insert data into the file. We have examples of both in our [Demo](https://github.com/OpenC3/cosmos/blob/main/openc3-cosmos-init/plugins/packages/openc3-cosmos-demo/targets/INST/cmd_tlm/inst_tlm.txt)

jurgensymynck, 2023-07-25:
Hi, thanks for the fast reply!

It turns out that my telemetry screen syntax and its ERB code is probably not to blame, I narrowed it down to a possible parsing bug. see issue #755.

Apart from the specific bug I was chasing, can you answer the original question? That is: is there a log file from the parsing of the plugin files that I could have checked for the occurrence of - for example - _Unknown widget: <%Widget_ ?

Thanks!
grts, Jurgen

jmthomas, 2023-07-25:
Plugin parsing errors should be visible in the Admin -> Plugin page when a plugin is installed. Plugin installation statuses are removed after 10min. Click the eye icon to view them.