OpenC3/cosmos #754: Can't save Telemetry Viewer screen in edit mode

finchlt, 2023-07-25:
**Describe the bug**
Save button for Telemetry Viewer in edit mode does not seem to work.

**To Reproduce**
Steps to reproduce the behavior:

1. Go to 'Telemetry Viewer'
2. Load valid screen
3. Open screen edit dialog
4. Make update
5. Hit save
6. Nothing happens

**Expected behavior**
Telemetry screen edit dialog save updates and exits

- OS: Linux (Ubuntu)
- OpenC3 COSMOS Version 5.9.1
- Browser
  -- [] Mozilla Firefox

jmthomas, 2023-07-25:
I'm not able to reproduce in Chrome or Firefox on Mac OS. Can you open your browser console and see if there are any errors thrown?

finchlt, 2023-07-25:
I had a small error in config, but it gave me no indication there was an error. Fails silently

jmthomas, 2023-07-25:
Can you create a reproducible example?

ryanmelt, 2023-07-27:
Can you give an example of the type of config error you had?

finchlt, 2023-07-28:
I was a VERTICAL or HORIZONTAL section that was missing a END I think

jmthomas, 2023-12-14:
Unable to reproduce.

closed 2023-12-14 by jmthomas