OpenC3/cosmos #755: Possible parsing bug while parsing specific comment string # (-): in telemetry screen definition file

jurgensymynck, 2023-07-25:
**Describe the bug**
It looks like the telemetry screen parser breaks when trying to parse a specific comment line `# (-):` in screen definition files.

**To Reproduce**
Write a telemetry screen definition file like the lower minimum working example, add to a plugin, upload and activate the screen.
Telemetry screen throws an _Error: Unknown widget: <%Widget_

```
SCREEN AUTO AUTO 0.2

VERTICAL
  # (-):
  <% (0..2).each do |i| %>
    BUTTON 'ON' 'api.cmd("<%= tisco_target_name %> CMD_SET_JOINT_POWER_ON with JOINT_NUMBER <%= i %>")'
  <% end %>
END
```

... while the following file is correctly parsed and yields the following screen. Note the absence of `# (-):`

```
SCREEN AUTO AUTO 0.2

VERTICAL
  #
  <% (0..2).each do |i| %>
    BUTTON 'ON' 'api.cmd("<%= tisco_target_name %> CMD_SET_JOINT_POWER_ON with JOINT_NUMBER <%= i %>")'
  <% end %>
END
```

**Environment:**

- Windows 10
- Google Chrome Versie 115.0.5790.102 (Officiële build) (64-bits)
- OpenC3 COSMOS 5.9.1
- Docker Desktop v4.21.1
- Ruby 3.2.2 (2023-03-30 revision e51014f9c0) [x64-mingw-ucrt]

jmthomas, 2023-07-25:
I think the confusion is that we only do ERB parsing on plugin installation. If you edit a file and add ERB syntax `<%` it will fail because that is not valid syntax. Is that the issue here?

jurgensymynck, 2023-07-26:
Hi,

No, I remember that we discussed this already some issues ago, like in 2022.

I narrowed it down further: the comment section that is problematic is `# (—):` and working comment section is `# (-):`
Spot the difference? It's a matter of a few pixels... The problematic character is an "em-dash". 

To complicate things I thoughtlessly replaced the em-dashes with a minus sign in my bug report. Today I realized that this special character might be the cause, and it was. 

I reintroduced the em-dash in my original post title and body, if you check this updated version you hopefully can reproduce.

For full context, this was my original  comment block:
```
    # LABEL "+ : step servos clockwise (CW)"
    # see https://www.techtoolsforwriters.com/how-to-create-an-em-dash/
    # Em dash (—): Alt+0151
    # En dash (–): Alt+0150
    # LABEL "- : step servos counterclockwise (CCW)"
```
So, my problem is solved more or less, I know the offending character. 

Now, about the parsing error: A quick search in https://docs.openc3.com/docs/v5/format doesn't reveal any limitations on the file encoding to be used for configuration files. Mine is UTF-8.

greetings, Jurgen

jurgensymynck, 2023-07-26:
... and for completeness,  both   
- Em-dash (—): Alt+0151
- En dash (–): Alt+0150
 
... give parsing errors

Never thought that successfull debugging can be a matter of spotting two extra pixels (i counted them) ...

[image]

I am going to have to renew my glasses prescription!  :-)

gr. Jurgen

jmthomas, 2023-07-26:
Glad you were able to find it! We've had similar issues in the past with "special quotes" that Microsoft likes to create from basic quotes during a copy and paste. I thought we had solved that with the encoding stuff. Thanks for your report!

ryanmelt, 2023-07-26:
A UTF-8 character in a comment should not break screen definition files.

jurgensymynck, 2023-08-03:
Hi,

I want to add to the bug description that the error messages caused by this bug are quite useless and misleading.
I had a similar problem in my tlm.txt config file: 
```
TELEMETRY <%= tisco_target_name %>  [...]
  <% number_of_joints.to_i.times do |i| %>
    APPEND_ITEM [...]
    ...
    UNITS milli-Ampère mA
    ...
  <% end %>
```

A workaround is to use plain full unit names:
```
UNITS milliAmps mA
# UNITS milli-Ampère mA
```
but, as we know, also the comment string itself break the parser. The error message in both cases (abriged) is below, which is quite useless. My latest problems (issue #762, 752) stem mostly from this lack of accurate reporting.

- undefined **local variable** or method `tisco_target_name' 
- **Unknown keyword** [...] for Target: MAP_TARGET TISCO
- **Unknown keyword** [...] for Target: PROTOCOL READ_WRITE

greetings, Jurgen

```
RuntimeError : 
Error processing [...]tlm.txt:
ERB error at [...]tlm.txt:7
undefined local variable or method `tisco_target_name' for #<OpenC3::ConfigParser:0x00007efe899d1640 @url="https://openc3.com/docs/v5", @target_name="TISCO", @filename="[...]tlm.txt">

plugin.txt20230803-626-bad3df:17: MAP_TARGET TISCO
Error: Unknown keyword and parameters for Target: MAP_TARGET TISCO

plugin.txt20230803-626-bad3df:19: PROTOCOL READ_WRITE LengthProtocol 40 8 0 1 LITTLE_ENDIAN 0 0x555555D5 nil true nil
Error: Unknown keyword and parameters for Target: PROTOCOL READ_WRITE LengthProtocol 40 8 0 1 LITTLE_ENDIAN 0 0x555555D5 nil true nil
```

jmthomas, 2023-12-14:
closed by #775

closed 2023-12-14 by jmthomas