OpenC3/cosmos #757: stop_callback for Script Runner Suites

jmthomas, 2023-07-26:
It would be nice to have a `stop_callback` method which would be called when a user clicks Stop in a Suite. This can be used to cleanup any current operations in response to Stop.

jmthomas, 2023-07-26:
@ryanmelt I looked into it and i think the place is script/suite.rb:380 where we handle StopScript. The only problem is calling a method here doesn't work due to running_script.rb:797 in `pre_line_instrumentation`which does `raise OpenC3::StopScript if @stop`. Assigning to you in case you have any ideas how to best implement this.

ryanmelt, 2023-09-05:
I think we can use standard Ruby at_exit for this.   Need to research to see if there is an equivalent for Python.

ryanmelt, 2023-09-12:
After experimentation, it would be better to add the ability to register our own callbacks, such that we can run the functions before the script has officially stopped and cleaned up