OpenC3/cosmos #762: How to use variables in Ruby ERB (<% %>): Global, Local and Instance variables, examples needed

jurgensymynck, 2023-07-27:
hello,

Being a complete Ruby noob, I am having trouble interpreting [this entry about VARIABLES](https://docs.openc3.com/docs/v5/plugins#variable), which states:

> The VARIABLE keyword defines a variable that will be requested for the user to enter during plugin installation.[...] **The variables are assigned to accessible local variables in the file**. At a high level, ERB allows you to run Ruby code in configuration files.

I read the Ruby Guide [as suggested here](https://docs.openc3.com/docs/v5/scripting) in the COSMOS scripting guide for info about local variables. I came across [this entry about variables.](https://ruby-doc.org/docs/ruby-doc-bundle/Manual/man-1.4/syntax.html#variable)
It speaks of these types:
- Global variables, Example:	$foobar
- Local variables, Examples:	foobar
- Instance variables, Example: @foobar

Alas, a telemetry screen definition file with any of these lines fails to run properly. I tried all sorts of combinations to no avail:

in plugin.txt:
```VARIABLE my_var Foobar```

in my_telemetry_screen.txt:
```
LABEL NotMyFooBar

LABEL "#{$my_var}"
LABEL "#{@my_var}"
LABEL "#{my_var}"

LABEL "<%= $my_var %>"
LABEL "<%= @my_var %>"
LABEL "<%= my_var %>"

LABEL <%= $my_var %>
LABEL <%= @my_var %>
LABEL <%= my_var %>
```

Using Githubs search function, I tried to find examples in the COSMOS demo plugin of usage of these types but i came up short...
- [search $ sign in .TXT files](https://github.com/search?q=path%3A*.txt+repo%3AOpenC3%2Fcosmos+%24&type=code)
- [search @ sign in .TXT files](https://github.com/search?q=path%3A*.txt+repo%3AOpenC3%2Fcosmos+%40&type=code)
 
Can you help my with the syntax or point me to example usages in thee COSMOS demo?

Thanks!
gr. Jurgen
***
Ultimately, I try to create a for-loop with following pseudo code:

plugin.txt:
```
VARIABLE tisco_target_name TISCO
VARIABLE number_of_joints 8
```

my_telemetry_screen.txt:
```
<% (0..$number_of_joints).each do |i| %>
    # do something for each joint
    LABELVALUE <%= tisco_target_name %> MY_TLM_PACKET TLM_VALUE_<%= i %>
<% end %>
```

ryanmelt, 2023-07-27:
Your example looks good, except number_of_joints should look like a local variable.  (and ruby is range inclusive so you should probably each times instead of a range)

```
<% number_of_joints.times do |i| %>
    # do something for each joint
    LABELVALUE <%= tisco_target_name %> MY_TLM_PACKET TLM_VALUE_<%= i %>
<% end %>
```

ryanmelt, 2023-07-27:
Note that ERB is only applied at plugin install time.   Doing this directly in the built-in screen editor won't work.

jmthomas, 2023-07-27:
I think you're confusing VARIABLEs, which are used in plugin.txt for installing plugins, with Ruby variables for scripting. They are separate concepts. A VARIABLE declared in a plugin is only available to that plugin when it is installed. It is primarily used to make your plugin more flexible to the people using it by creating user configurable items like IP addresses, ports, etc.

Like Ryan said, ERB code is only evaluated during installation time and thus is most suited to generating target names, creating templates to create multiple items, etc. Once COSMOS is up and running you'll notice when you edit a screen that the ERB templating disappears and the actual values are filled in.

Variables in Ruby scripts can be creating using all the features of the Ruby programming language which includes globals (prefixed with `$`, instance variables (prefixed with `@`), and class variables (prefixed with `@@`). Of course your usage of these completely depends on how you structure your scripts.

jurgensymynck, 2023-08-03:
Hello,

I finally got things working as intended, slightly modifying the suggestion of  https://github.com/OpenC3/cosmos/issues/762#issuecomment-1654532946, because by default the loop apparently treats the VARIABLE _number_of_joints_ as a text string: ```<% number_of_joints.to_i.times do |i| %>```

Thanks! 
gr. Jurgen

--- 

Solution:
plugin.txt:
```
VARIABLE tisco_target_name TISCO
VARIABLE number_of_joints 3
```
_tlm_common_header.txt (just for completeness)
```
#<%# @file _tlm_common_header %>
    APPEND_ITEM SYNC_PATTERN 32 BLOCK "Sync Pattern" LITTLE_ENDIAN
    APPEND_ID_ITEM PACKET_ID 8 UINT <%= packet_id %>
      FORMAT_STRING "0x%0X"
      APPEND_ITEM FRAME_SIZE 8 UINT "Frame size (bytes)"
      UNITS Bytes B
    APPEND_ITEM TIMESTAMP 32 UINT "Time in milliseconds [ms] since TiSCo startup"
      UNITS millis ms
```
tlm.txt
``` 
...
TELEMETRY <%= tisco_target_name %> TLM_JOINTPWMPULSE LITTLE_ENDIAN
  <%= render "_tlm_common_header.txt", locals: {packet_id: 0xA1} %>
  <% number_of_joints.to_i.times do |i| %>
    APPEND_ITEM JOINTPWMPULSE_<%= i %> 16 UINT "Joint/Servo [<%= i %>]"
  <% end %>
...
```

closed 2023-08-03 by jurgensymynck