OpenC3/cosmos #773: Error 500 when attempting to update plugin

apogeecmb, 2023-08-10:
**Describe the bug**
User gets a 500 error when trying to upgrade a plugin.  After selecting the gem file and clicking OK in the confirmation window, a 500 error banner is displayed at the top of OpenC3 and the plugin update fails.  Going into the browser inspector and finding the error in the console, we are seeing the following error message details:  {"status":"error","message":"Token is required"}.  The only way currently we've been able to recover from this scenario is to completely stop and then restart OpenC3.  

**To Reproduce**
Steps to reproduce the behavior:

1. Start OpenC3
2. Wait some unknown length of time.  We don't see this problem immediately, but only some time later (hours or days) after OpenC3 has been running.
3. Attempt to update a plugin.
4. See 500 error.

**Expected behavior**
The expected behavior is for the plugin update to succeed.  I'm guessing the token in question is some kind of authorization token.  If the token has expired, I would expect OpenC3 to force the user to re-login or something of that nature to get a new token.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

- OS: Windows 10
- OpenC3 COSMOS Version - 5.5.1 (open source edition)
- Browser
  -- [] Internet Explorer (IE)
  -- [X] Google Chrome
  -- [X] Chromium based browsers
  -- [] Mozilla Firefox
  -- [] Safari

ryanmelt, 2023-08-11:
Can you try with our latest release?

apogeecmb, 2023-08-11:
We definitely can.  Was there something specific that's been changed that you think might resolve the problem?  We were planning to update soon anyway to use the new python support, but we're in the middle of doing a release, so we were holding off at the moment until that was completed.

ryanmelt, 2023-08-11:
Not particularly, but we added some better debugging in 5.5.2.    
5.5.1 was 6 months ago and this is the first I've heard of an upgrade problem, so hopefully it has already been addressed.

apogeecmb, 2023-08-15:
After updating to 5.10.0, we're still seeing the same error occur.  Here's a screenshot.  
[image]

We currently have two developers working independently (from two different PCs and browsers) but using the same instance of OpenC3.  Is it possible since there's just a single login credential that whatever authentication token is being issued is getting out of sync or something with the two users logging in separately?  I updated to 5.10.0 this past Friday and didn't see this error myself.  However another user logged in yesterday (Monday) and started working.  She then had the error occur and subsequently now it's there for everyone who tries.

apogeecmb, 2023-08-15:
Another update:  We were able to do some more troubleshooting and nailed down the specific cause of the problem.  The issue occurs when a different user starts OpenC3.  Since the openc3.sh script pulls the current user's id when it starts the Openc3 containers, any plugin created in OpenC3 is owned by that user.  Subsequently, if a different user starts OpenC3, they won't have permissions to overwrite the existing plugin file when an update is attempted.

Internally we're going to resolve this by creating a new "openc3" user on our system that will always be used to start and stop OpenC3.  That should resolve any file permissions issues within the containers.  Maybe this was the intended use case of the developers all along?  Just thought I'd pass this info along in case you feel you want to make any changes on your side.

ryanmelt, 2023-08-15:
Thanks for the update.  File permissions can be tricky.  We run as the same user id as the host user to ensure that we can write to host mounted folders (like the plugins folder for local mode).   It was assumed that the same user would be starting COSMOS each time.

I'll leave this ticket open for a few days to do some research.  We might be able to add an environment variable to control the permissions put on written files.  By default it would probably stay as is (only user writable), but we could potentially open that up to group writable, or world writable to support a shared installation between multiple users.

ryanmelt, 2023-08-15:
Note for myself: Also investigate using umask

tmzhang2023, 2023-11-03:
We encountered the same error message when upgrading plugin via the web browser.

Setup:
AWS cloud instance with Ubuntu (20.04.6 LTS (Focal Fossa))
OpenC3 COSMOS Open Source 5.11.3
Firefox 119.0 (64-bit)
Chrome 118.0.5993.117 (Official Build) (64-bit)

Issue description:
Starting with gem file version 1.0.3, the "HTTP 500" error message appears at the top of the browser when uploading the gem file even though it was generated successfully. Gem files version 1.0.0, 1.0.1 and 1.0.2 were all generated, uploaded and activated successfully with no issues. This error message occurs on both Firefox and Chrome.

A group account was used to log into the instance and perform all operations required to setup and configure COSMOS.

On the online documentation webpage "[Getting Started | OpenC3 Docs](https://docs.openc3.com/docs/getting-started/gettingstarted)", there is a "BROWSER VERSION ISSUE" section that talks about browser compliance. 
But we are having the "HTTP 500" issue with both Firefox and Chrome, so what is the recommended web browser and version? Also, what is the recommended latest stable release of COSMOS?

Does anyone know the cause of this problem? If there is a workaround, please let us know.

Thanks so much!

ryanmelt, 2023-11-04:
Most likely there is a problem in the content of your plugin file.    You can get more info by doing "docker logs <cmd-tlm-api container hash>"

tmzhang2023, 2023-11-06:
Thanks Ryan! I reverted the plugin file back to the version when it was successfully loaded and got the same error, but I will look into the docker logs for clues.

ryanmelt, 2024-01-07:
COSMOS must always be started and run by the same user account or you will have permissions issues.

closed 2024-01-07 by ryanmelt