OpenC3/cosmos #776: When commenting out a line with '#', also comment following ERB code '<%#=...%>'

jurgensymynck, 2023-08-13:
**Is your feature request related to a problem? Please describe.**

Commenting out command , telemetry or screen definitions by just placing a ``#`` symbol in front of a line containing Ruby ERB `<% ... %>`has caused me some headaches , trying to decipher unrelated error messages like "Warning TLM_XXX redefined and "Error: packet_id undefined" 
this example doesn't work:
```
TELEMETRY <%= tisco_target_name %> TLM_PING LITTLE_ENDIAN
  <%= render "_tlm_common_header.txt", locals: {packet_id: 0x81} %>

# As of now, it's too soon to experiment with BLOCK data type...
# TELEMETRY <%= tisco_target_name %> TLM_ECHO_BLOCK LITTLE_ENDIAN
#   <%= render "_tlm_common_header.txt", locals: {packet_id: 0x9E} %>
#   APPEND_ITEM DATA_0 8 BLOCK "Data 0"

TELEMETRY <%= tisco_target_name %> TLM_ECHO_UINT32 LITTLE_ENDIAN
  <%= render "_tlm_common_header.txt", locals: {packet_id: 0x8E} %>
  APPEND_ITEM DATA_0 8 UINT "Data_0"
    FORMAT_STRING "0x%0X"
```
... but this works:
```
TELEMETRY <%= tisco_target_name %> TLM_PING LITTLE_ENDIAN
  <%= render "_tlm_common_header.txt", locals: {packet_id: 0x81} %>

# As of now, it's too soon to experiment with BLOCK data type...
# TELEMETRY <%#= tisco_target_name %> TLM_ECHO_BLOCK LITTLE_ENDIAN
#   <%#= render "_tlm_common_header.txt", locals: {packet_id: 0x9E} %>
#   APPEND_ITEM DATA_0 8 BLOCK "Data 0"

TELEMETRY <%= tisco_target_name %> TLM_ECHO_UINT32 LITTLE_ENDIAN
  <%= render "_tlm_common_header.txt", locals: {packet_id: 0x8E} %>
  APPEND_ITEM DATA_0 8 UINT "Data_0"
    FORMAT_STRING "0x%0X"
```

**Describe the solution you'd like**
- When using the built-in OpenC3 Cosmos editor, when applying a comment in all Cosmos screens (plugin.txt, bridge.txt telemetry, command, script runner screens ...), using the normal method (on my system that's `<CTRL> - :`, all Ruby ERB code on that line should also be commented out, from `<%= ...` to `<%#= ...`
- The above behavior should be enabled/disabled in some settings window.
- Eventually, this behavior should be implemented in a new COSMOS VS Studio Code extension (see #756) so the same thing happens  while local editing (I will check if such a extension with this functionality ready exists)

gr. Jurgen

tj-scherer, 2023-08-16:
I have run into this issue before. The ERB ignores comments, and at the time there was no interest in changing the behavior. See #60 for reference. I believe this behavior is unintuitive and should be addressed, since it won't affect backwards compatibility, and will save many headaches as you mentioned.

jmthomas, 2023-08-16:
Unfortunately this is just how ERB works when it processes a file. As @jurgensymynck figured out, you can comment out ERB statements with `<%#` but that's both unintuitive and annoying when your editor simply comments out the entire line. I think the only way for us to handle this in COSMOS would be to pre-process the file and strip out any comment lines before running ERB. I'll look into that as a possible solution.

PR #820 (2023-09-13): Comment out ERB
closes #776
closed 2023-09-13 by jmthomas