OpenC3/cosmos #778: Ordering of protocols

tj-scherer, 2023-08-14:
When you define the protocols for an interface, it is specified that ["Protocols can be layered and will be processed in order"](https://docs.openc3.com/docs/v5/protocols).

I have 2 custom protocols:
1. TRUNCATE packet using `write_data()`
2. LENGTH of packet using `write_data()`

My plugin specifies the TRUNCATE protocol first, then the LENGTH protocol, similar to below:
```
TARGET ...
INTERFACE ... 10.0 nil
  MAP_TARGET ...
  PROTOCOL READ_WRITE <TRUNCATE>
  PROTOCOL READ_WRITE <LENGTH>
```
In this scenario, the LENGTH `write_data()` gets called before the TRUNCATE `write_data()`, which I can see because the packet getting sent contains the length before TRUNCATE runs, but the actual length of the packet reflects what TRUNCATE does.

I would expect the opposite order to be used, or this behavior to be documented.

- OS: Ubuntu 20.04
- OpenC3 COSMOS Version 5.9.1

ryanmelt, 2023-08-14:
It is documented, though probably not well enough:
https://docs.openc3.com/docs/v5/protocols

Search for "reverse".

The write protocols run in the reverse order of the read protocols.

tj-scherer, 2023-08-14:
Ah that's what I was looking for, thank you! Perhaps that information would be worth putting in one of the highlighted blocks such as this?

[image]