OpenC3/cosmos #783: 'openc3-cosmos-network already exists' error response

jimhar-mn, 2023-08-17:
Hello,

Looking into Cosmos 5 after using Cosmos 4 in the past.

During installation and plug-in build steps I repeatedly seem to get the following error:

`Error response from daemon: network with name openc3-cosmos-network already exists`

Everything seems to go fine after that message, but I was just wondering what it meant.

Cosmos 5.10.0.  Windows 10.

ryanmelt, 2023-08-17:
Thanks for checking out COSMOS 5!

That's a Docker error message that happens when you try to create a docker network that already exists.    We try to create that network in several places to make sure it exists.   The error can be safely ignored.

ryanmelt, 2023-09-11:
On windows it looks like we can pipe the output to NUL

ryanmelt, 2024-10-13:
We no longer create the network and let docker compose handle it.

closed 2024-10-13 by ryanmelt