OpenC3/cosmos #787: COSMOS Hangs at Launch

Gunblade1337, 2023-08-18:
I am running Linux Mint on VirtualBox and followed the guide to install Docker Desktop and have that successfully running.  I've checked out the COSMOS tutorial repo and when I run 
`./openc3.sh run`
The only line that gets printed out is
`Docker Compose version V2.20.2-desktop.1`
and the program hangs there indefinitely.  I did follow this instruction here during the Docker setup:
`If you use an Ubuntu derivative distro, such as Linux Mint, you may need to use UBUNTU_CODENAME instead of VERSION_CODENAME.`
I have Docker Desktop and COSMOS running successfully on the Windows side (Win 10 Home) so I know what I should be seeing.  Has anyone experienced this issue before?

Gunblade1337, 2023-08-18:
I just tried again using Ubuntu 22.04.3 LTS on VirtualBox but I get the same result.

ryanmelt, 2023-08-19:
Sounds like you may be nesting virtualization too deep.  
On linux you should generally use docker directly, not use Docker Desktop.
What is your host OS?

Gunblade1337, 2023-08-19:
Host OS is Windows 10 Home edition. I was able to get Docker Desktop working there by enabling hypervisor from the command line:
_bcdedit /set hypervisorlaunchtype auto_

I don't know how the host OS affects my VM so I've tried running Cosmos inside the VM with both hypervisor off and on. However, with your suggestion I can try Cosmos without Docker Desktop.

Gunblade1337, 2023-08-19:
Looks like I got it working with just Docker Engine! Thank you Ryan.

closed 2023-08-19 by Gunblade1337