OpenC3/cosmos #796: Telemetry Grapher backend disconnected

skliper, 2023-08-28:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Describe the bug**
Manually selecting an item to graph or opening graph on a live value in Packet Viewer results in an error:
```
        "type": "disconnected",
        "message": "OpenC3 backend connection disconnected",
        "time": 1693256021550
```

Might be similar to #559?

**To Reproduce**
I've tried stopping/starting all the containers, even removed all the volumes.  The target is connected, cmd server says I'm receiving data, the tlm packets for the value I'm looking at are incrementing.

**Expected behavior**
A graph of the data

**Screenshots**
If applicable, add screenshots to help explain your problem.

```
>docker ps
CONTAINER ID   IMAGE                                                                  COMMAND                  CREATED          STATUS          PORTS                                             NAMES
00b9cfb1de6e   openc3inc/openc3-cosmos-cmd-tlm-api:latest                             "/sbin/tini -- rails…"   20 minutes ago   Up 20 minutes   2901/tcp                                          cosmos-project-openc3-cosmos-cmd-tlm-api-1
aa5bf543e55f   openc3inc/openc3-operator:latest                                       "/sbin/tini -- ruby …"   20 minutes ago   Up 20 minutes   0.0.0.0:1235-1237->1235-1237/udp                  cosmos-project-openc3-operator-1
c5559485f461   openc3inc/openc3-traefik:latest                                        "/entrypoint.sh trae…"   20 minutes ago   Up 20 minutes   127.0.0.1:2900->80/tcp, 127.0.0.1:2943->443/tcp   cosmos-project-openc3-traefik-1
b9fb3dc2a593   openc3inc/openc3-cosmos-script-runner-api:latest                       "/sbin/tini -- rails…"   20 minutes ago   Up 20 minutes   2902/tcp                                          cosmos-project-openc3-cosmos-script-runner-api-1
f6ff146e3937   openc3inc/openc3-minio:latest                                          "/usr/bin/docker-ent…"   20 minutes ago   Up 20 minutes   9000/tcp                                          cosmos-project-openc3-minio-1
29e1ea4fafcf   openc3inc/openc3-redis:latest                                          "docker-entrypoint.s…"   20 minutes ago   Up 20 minutes   3680/tcp, 6379/tcp                                cosmos-project-openc3-redis-1
5f89a4f56d5f   openc3inc/openc3-redis:latest                                          "docker-entrypoint.s…"   20 minutes ago   Up 20 minutes   3680/tcp, 6379/tcp                                cosmos-project-openc3-redis-ephemeral-1
196ed8b6b1fb   aetd-dockerlab.gsfc.nasa.gov/gsfc-cfs/user_tools/docker/cfs-ubuntu22   "/bin/bash"              3 months ago     Up 9 hours                                                        cfs-ubuntu22-c
```

**Environment (please complete the following information):**

- OS: Windows 10
- OpenC3 COSMOS Version: 5.10.0
- Browser
  -- [] Internet Explorer (IE)
  -- [x] Google Chrome
  -- [] Chromium based browsers
  -- [] Mozilla Firefox
  -- [] Safari

ryanmelt, 2023-08-28:
Try graphing and then running "docker logs 00b9" or whatever the correct hash is for the cmd-tlm-api container.

skliper, 2023-08-29:
Oh fun, looks like a problem w/ the authorization... note I've changed the auth since I'm guessing I shouldn't expose it publicly.  Seems like `%` is a problematic char?  Is it not getting quoted correctly?

```
I, [2023-08-29T13:37:01.779300 #282]  INFO -- : [dc35ddbb-c073-4c31-b671-6018d29b5ccb] Started GET "/openc3-api/cable?scope=DEFAULT&authorization=XYZ%ZYX" for 172.19.0.1 at 2023-08-29 13:37:01 +0000
E, [2023-08-29T13:37:01.780406 #282] ERROR -- : There was an exception - ActionController::BadRequest(Invalid query parameters: invalid %-encoding (XYZ%ZYX))
E, [2023-08-29T13:37:01.780455 #282] ERROR -- : /usr/lib/ruby/3.1.0/uri/common.rb:370:in `_decode_uri_component'
```

skliper, 2023-08-29:
Update - removed special characters from my initial password and it worked.

PR #818 (2023-09-13): encodeURI for action cable consumers
closes #796
closed 2023-09-13 by jmthomas