OpenC3/cosmos #798: Add ability to list installed plugins to openc3cli

ryanmelt, 2023-08-30:


bradenvirdellgenevatech, 2024-04-15:
Hello Ryan

Just want to say this enhancement would be really useful for our current situation and was wondering where it is in your priority list.

bradenvirdellgenevatech, 2024-04-15:
@jmthomas 
That's exactly what I am looking for, I suspect it can be used with the ./openc3.sh script as well like the load and unload can be?

jmthomas, 2024-04-15:
Yep just another option to the `openc3.sh cli`. It'll be `openc3.sh cli list`.

bradenvirdellgenevatech, 2024-04-16:
Prefect, thank you four the quick response, cant wait for the next release.

bradenvirdellgenevatech, 2024-04-18:
@jmthomas Just thought of something regarding this. does it list all installed gems across all scopes? If it does it would be nice to have an easy way to tell them apart. Either listing out the scope with the gem or maybe having an argument for the command to ask for which scope you would like to list

closed 2024-04-29 by jmthomas