OpenC3/cosmos #804: Having trouble with the JSON Accessor and packet DECOM

GTI-Dakota, 2023-09-01:
Hello,

I am having issues with the JsonAccessor with some of my packets. The Decom microservice cannot process the json string I return from my protocol. I see this happening in Arrays of strings, and values with colon characters in them. I have referenced your packet spec found here: 
(https://github.com/OpenC3/cosmos/blob/9346547e296523cf98a80eca21f7d848e61705d4/openc3/spec/packets/packet_spec.rb#L1643)

List packet definition:
```
TELEMETRY <%= target_name %> TLM LITTLE_ENDIAN ""
	ACCESSOR JsonAccessor
	TEMPLATE {"tlm1":0,"tlm2":"[\"list\"]","tlm3":""}
	APPEND_ITEM TLM1 64 INT "telemetry 1"
		KEY $.tlm1
	APPEND_ARRAY_ITEM TLM2 256 STRING 2560  "telemetry 2"
		KEY $.tlm2
	APPEND_ITEM TLM3 2048 STRING  "telemetry 3"
		KEY $.tlm3
```
My JSON packet for this list looks like: `{"tlm1":4,"tlm2":["LIST"],"tlm3":"TEST"}`

Colon packet definition:
```
TELEMETRY <%= target_name %> TLM_B LITTLE_ENDIAN ""
	ACCESSOR JsonAccessor
	TEMPLATE {"tlm1":"","tlm2":""}
	APPEND_ITEM TLM1 256 STRING "telemetry 1"
		KEY $.tlm1
	APPEND_ITEM TLM2 2048 STRING "telemetry 2"
		KEY $.tlm2
```
My JSON packet with a colon looks like: `{"tlm1":"cmd:test:cmd1","tlm2":"TEST"}`

**Expected behavior**
I do not expect to be having issues with these packets, as many other packets come in with no issue

**Screenshots**
Screenshot of decom error (it is the same for both types of packets)
[image]

**Environment (please complete the following information):**
- OS: Ubuntu 20.04
- OpenC3 COSMOS Version 5.8.1
- Browser
  -- [x] Internet Explorer (IE)
  -- [x] Google Chrome
  -- [x] Mozilla Firefox

Thank you!

Dakota

ryanmelt, 2023-09-01:
Unexpected token at '' generally means that the buffer is empty
If you try JSON.parse('') you will get the exact same error

Are you sure the telemetry packet received contained the JSON data?

GTI-Dakota, 2023-09-01:
Yep, here are the two code variants (in my protocol) I just used to confirm that the packet.buffer is not empty:

```
packet.buffer = new_buffer.to_json
Logger.info(new_buffer.to_json)
```
            
```
packet.buffer = new_buffer.to_json
Logger.info(packet.buffer)
```

ryanmelt, 2023-09-01:
The exception is happening at the decom_microservice though... I'm not sure where you added the debug printout.
Can you look at the corresponding raw_log file for the same time this occurs?

GTI-Dakota, 2023-09-01:
The code above is in read_packet right before `return packet`:
[image]
There are no more protocols that follow this one.

Could I email you the raw log file? I think I found it

ryanmelt, 2023-09-01:
Yes please

ryanmelt, 2023-09-03:
This problem was caused by the buffer data containing a large number of 0 bytes at the end of the string.   The JSON parser does not handle zero bytes, and those need to be stripped.

ryanmelt, 2023-09-12:
closed by #805

closed 2023-09-12 by ryanmelt