OpenC3/cosmos #813: Screens not opening from script runner

djustice-gravitics, 2023-09-06:
I am having issues with screens showing when using script runner.  I am trying to run a basic script:

temp = 0 # This variable is accessed in the screen
screen_def = '
  SCREEN AUTO AUTO 0.1 FIXED
  VERTICAL
    TITLE "Local Variable"
    VERTICALBOX
      LABELVALUE LOCAL LOCAL temp # Note LOCAL LOCAL
    END
  END
'
# Here we pass in the screen definition as a string
screen = local_screen("My Screen", screen_def, 100, 100)
500.times do
    temp += 1 # Increment temp to update the screen
end

The script runs without issue but nothing displays.  I have tried other examples and the screens never display.  

Ubuntu 22.04.01
Firefox 116.0.3

ryanmelt, 2023-09-06:
Looks like some of this is a leftover from COSMOS 4.   

You cannot use local variables in local screens in COSMOS 5, because the screen is running in the browser, and not in the same application as the script which is running in the background.

I will update the documentation, to reflect that.   
If you do any normal screen LABELVALUE to show telemetry from packets, that will work.

ryanmelt, 2023-09-06:
See:
https://github.com/OpenC3/cosmos/blob/main/openc3-cosmos-init/plugins/packages/openc3-cosmos-demo/targets/INST/procedures/screens.rb

For a working example of a local screen

djustice-gravitics, 2023-09-06:
Thanks I was able to get an example to work.

closed 2023-09-06 by djustice-gravitics